JuliaControl/ControlSystems.jl #1: Add tests for current functionality

jcrist, 2015-01-20:
Testing is woefully incomplete. This should be fixed before new features are added.

jcrist, 2015-02-10:
Still need to figure out how to test time/frequency domain analysis. These generate large arrays of data, and would be hard to test with the results stored in the test file. One option is to save the arrays to a binary file (ascii may also work, but is loss-y), and then compare if the result is approximately the same using `@test_appx_eq`.

Also need to figure out how to test plots. A test for lack of error (run the function to ensure it doesn't error out), along with a data comparison for the result data may be enough.

Besides the above, everything currently has testing working.

mfalt, 2016-05-02:
The test coverage is continuing to grow and once we pull #55 we will have 66% coverage.
The plots are tested by saving reference images in the `ControlExamplePlots.jl` repo and doing an automatic comparison using `VisualRegressionTests.jl`.
If the error is too large the test will not pass. The images can be updated by manually running the "runplotgen()" in ControlExamplePlots.jl.
It is not an optimal solution since it will be tricky to have several branches with very different plotting pass the tests at the same time. But it also keeps the size of the toolbox down (ControlSystems.jl ≈ 113KB compared to ControlExamplePlots ≈ 338KB), and keeps the version history cleaner.
It does add a significant time to the tests, but I believe that it is worth it to have reliable plotting (even if we are currently only checking one back-end).

closed 2020-10-10 by baggepinnen