JuliaControl/ControlSystems.jl #2: Add docs

jcrist, 2015-01-20:
Add documentation through Read-the-docs. "Docstrings" are added using the `@doc` macro already, there may be a way to use these to pull these in to the documentation. 

It would be nice to have some examples as well. The `StateSpace` and `timeresp` stuff is already fairly functional, some simple examples from an intro linear-systems course could be added.

jcrist, 2015-02-10:
[This](http://michaelhatherly.github.io/Lexicon.jl/) can be used to scrape the docstrings into markdown.

phelipe, 2016-03-07:
Do you have any news about documentation?

mfalt, 2016-03-08:
There are a lot of big changes being pulled into this package in the next few days. Once this is done I will start looking into what package to use for documentation, any help is appreciated. I think a couple of good examples on how to use the different parts would go a long way.

baggepinnen, 2016-03-08:
I can probably contribute with the examples that motivated the implementation of the various methods

phelipe, 2016-03-08:
Thanks!  I can probably contribute to the documentation.

mfalt, 2016-03-15:
We should also be clear in the example and documentation on how the plotting works. If the user has PyPlot installed then the example works great, otherwise some other package will be used for plotting and the last line of the example should reflect this. The user could also start the example with for example `Plots.immerse()` to choose the `Immerse` back-end if installed, this should be stated somewhere.

mfalt, 2016-03-26:
We have now started to create a documentation website using the new package [Lapidary.jl](https://github.com/MichaelHatherly/Lapidary.jl) (the successor to Lexicon.jl and Docile.jl). All (almost) functions have docstrings that show up on the webpage and we have added a few examples (although a bit advanced) and a (very) short introduction. Anyone is free to contribute to any part, whether its docstrings, examples, the manual or ideas about the structure of the webpage.

The docs are structured and deployed by Lapidary and built by mkdocs. Any commit to master or a commit with a new tag will be built by travis and sent to the branch [gh-pages](https://github.com/JuliaControl/ControlSystems.jl/tree/gh-pages) for which the webpage can be viewed though (for example [http://juliacontrol.github.io/ControlSystems.jl/latest](http://juliacontrol.github.io/ControlSystems.jl/latest) ). If you want to build the docs locally (which is good for testing and verifying) there is a simple script docs/builddocs that builds the docs by running make.jl and mkdocs. Optionally it accepts the julia path, ex:
`./builddocs /my/julia/dir/julia`
This will take the markdown in docs/src and compile a html version in /docs/site.
The index for the documentation is structured in docs/mkdocs.yml and the main page in docs/src/index.md in which the other files are linked.

The example code that is contained in blocks like
```
```julia
tf(1)
tf(2)
...

# output

TransferFunction ...
...
```                  .
```
will be executed and the last output will be compared to the part after `# output`, this ensures that the docs are correct and current.
The plots that are used in the examples should be built in the docs/src/makeplots.jl code. These require that PyPlot is installed which takes a lot of time in travis and will therefore only be run when the docs are to be deployed. For more info on how to create the docs, see the docs for Lapidary.jl.

aytekinar, 2016-04-27:
@neveritt and I have looked at the readily available `.jl` files. Since we require a Julia version of at least v0.4, do you think we should also pay attention to the Julia documentation [suggestions][julia-docs]? Currently, we do not need to use the `@doc` macro as you have been writing the code documentation in `Markdown`. And the Julia team suggests that we should bundle examples as verbatim copies of the Julia REPL since they have future plans of implementing the so-called `doctests` that makes sure the code documentation itself is consistent with the actual implementation of the corresponding function/method. 

[julia-docs]: https://julia.readthedocs.org/en/release-0.4/manual/documentation/

mfalt, 2016-04-27:
I have not looked at that page before, but most if the points seem like good ideas, so feel free to change it, but I do not see it as a priority since the current way is working so far.
I am not convinced of using direct copies of the REPL though, I think there is a lot of value in being able to copy the code from the example directly into the REPL which breaks if the code contains `julia> ...`
Also, the doctest functionality is actually implemented in the documentation website, but without the `julia>` part.

aytekinar, 2016-04-27:
I think you are getting confused. I am not saying this for the `docs/` folder, which goes directly to readthedocs. You are definitely right in what you say --- if I am a student consulting the readthedocs webpage for the corresponding functionality, I will just copy the excerpt and paste in my Julia. However, the documentation we write just on top of the corresponding function needs to have `julia>` as per number 5 in the abovementioned official documentation page of JuliaLang.

mfalt, 2016-04-27:
No, I think I understand what you mean. But the documentation above the functions are available through the `?` command, and its useful to be able to copy the examples from that into the REPL. But you are right, it seems like it's not possible to get the doctest functionality then.

aytekinar, 2016-04-27:
As for now, I will write the documentation in the way JuliaLang team suggests, then, for the function `dcgain` for example.

In the future, if needed, we can create a branch to fix all the documentation accordingly?

mfalt, 2016-04-27:
All right, I think that is a decent solution.

closed 2020-10-10 by baggepinnen