JuliaControl/ControlSystems.jl #6: Add plotting functionality with Gadfly

jcrist, 2015-02-12:
Current plotting uses only PyPlot. It would be nice to add Gadfly as another plotting backend, as it only depends on Julia (not python).

One possible way of doing this is refactoring out the plotting commands (probably a good idea anyway) into separate function(s). This can then be included lazily using `Requires.jl`, with different files for `PyPlot` and `Gadfly`. A user is unlikely to load both in the same session, but if they do, then the last one loaded before a `*plot` function is called will be the one used. One way around this would be to include a `set_plot_backend(backend::Symbol)`, which sets a global, which is used in the plotting to determine which interface to call. This is similar to how the array printing works in base, with the `_limit_output` global. However, as stated above, it's unlikely that a user will be using two plotting packages at the same time, so this is probably unnecessary.

mfalt, 2016-03-03:
I think this should be implemented through the usage of the `Plots.jl`package, as started in #10
This allows the user to select back-end and keeps the code simpler.

mfalt, 2016-03-12:
Yes this is now implemented through 08b18c2, although not tested with Gadfly. The user can set the default backend to Gadfly by simply running
Plots.gadfly()
All plots in this package will then be done using gadfly. This should be documented.

closed 2016-03-12 by mfalt