JuliaControl/ControlSystems.jl #8: Mixed arguments in matrix_comps

JaUg3, 2015-10-12:
All functions in matrix_comps seems to be limited to accepting the same type of arguments, is this intended behavior?

For example, these work as expected
```
care(eye(1), eye(1), eye(1), eye(1))
```
```
e1 = eye(Int64,1)
care(e1, e1, e1, e1)
```
with output
```
1x1 Array{Float64,2}:
 2.41421
```
but you do not allow
```
care(eye(Int64,1), eye(1), eye(1), eye(1))
ERROR: MethodError: `care` has no method matching care(::Array{Int64,2}, ::Array{Float64,2}, ::Array{Float64,2}, ::Array{Float64,2})
```

I assume that this was a mistake, but since it is consistent over all functions in this file, as well as in a few others, I wanted to make sure before I spend any more time on it, especially since I am not familiar with the BLAS types. The solutions should be quite simple, mostly changing this:
```
function foo{T<:Bar}(a::T,b::T)..
```
to something like:
```
function foo(a::Bar,b::Bar)..
```
You are also not allowing scalars as inputs, is this intentional?

mfalt, 2016-03-03:
Fixed in 2083b75

jcrist, 2016-03-03:
Most of the algorithms that specify Float64s only work with floating points, and would probably lose precision with Float32. For ease of implementation I only implemented signatures with uniform types, but you could write them out in a general way, with a promotion to a consistent float size before computation.

jcrist, 2016-03-03:
Also, the referenced commit doesn't seem to fix this issue, I assume you mean https://github.com/mfalt/Control.jl/commit/c88932a8b7ddabddc629b8d86e39e1707246f6b9.

mfalt, 2016-03-03:
You are right regarding the commit, I don't know why I referenced the wrong one.
I think you are right that enforcing a high precision could be valuable, however the BlasFloat type already includes Float32, so this was not done before (we should however make sure that integers are treated properly). Another reason for not enforcing types is that several of these functions are so general that they should be able to handle a large range of types, for example symbolic SymPy types (with the exception `balance`). As I understand it, this is the preferred way to handle types in julia.

closed 2016-03-26 by mfalt