JuliaControl/ControlSystems.jl #11: rlocus must track roots

mfalt, 2016-03-11:
_From @baggepinnen on March 4, 2016 15:42_

the roots are sometimes mixed up in rlocus, there must be some intelligence to sort the roots such that the trajectories are continuous

_Copied from original issue: mfalt/Control.jl#32_

mfalt, 2016-03-11:
One alternative is to not draw lines, but simply plotting dots at each point, but this might not be a great solution. The best solution is probably to
```
for each K
   for each pole
        match the pole to the closest pole, in absolute distance at the previous k.
```
Something very similar to the minreal code.

mfalt, 2016-03-11:
_From @baggepinnen on March 4, 2016 19:49_

Dots do not solve the problem since the color is of importance. The for-loop approach deserves a try tomorrow

mfalt, 2016-03-11:
We should probably be careful with using `minreal` when calculating the poles though, or we might run into problems when the poles approach the zeros.

mfalt, 2016-03-11:
_From @baggepinnen on March 4, 2016 19:51_

Yes, I removed minreal from rlocus in last commit

mfalt, 2016-03-11:
We should probably do something like
```
P = minreal(P)
nump = numpoly(P)
denp = denpoly(P)
allroots = [roots(nump + k*denp) for k in K]
```
instead of using the `poles` function. (I invented the `numpoly` function here, so that `nump` and `denp` are simple polynomials)

baggepinnen, 2016-06-29:
I think there is a nice solution like this
- Differentiate the pole location with respect to the gain
- Integrate for root locus using ODE-solver with variable step size

This will automatically track the roots and take appropriate step sizes such that the root locus will be sufficiently detailed everywhere.

baggepinnen, 2017-08-30:
Adaptive stepsize solution used in #113
Automatically tracks roots

closed 2017-09-01 by baggepinnen