JuliaControl/ControlSystems.jl #15: gain is not working

mfalt, 2016-03-11:
_From @baggepinnen on February 23, 2016 20:2_

Should be renamed to static gain, or dcgain as in matlab.
Should also be ** working properly**

_Copied from original issue: mfalt/Control.jl#24_

mfalt, 2016-03-11:
Fixed for tf, ss remaining.

mfalt, 2016-03-12:
We can still not handle if the tf has both a pole and a zero in 0, this returns `NaN`.
The implementation for ss works better now, but can not handle when A is not invertible (for example with uncontrollable states). I think this could be solved using the psuedo-inverse. However, we still return a finite gain for unstable systems.

The previous implementation (where `nz=tzero(sys)`)
`sys.C*(sys.A^(nx - nz - 1))*sys.B`
still seems completely wrong.

aytekinar, 2016-04-27:
Assign the issue to me --- I am gonna fix this.

mfalt, 2016-04-27:
Great! @baggepinnen is currently implementing minreal for statespace which might be of use, but feel free to use some other way if it is more robust/efficient.

aytekinar, 2016-04-27:
I have already forked the repository, and I will create a branch `15-dcgain-fix`. In the functions I will not assume pole-zero cancellations (as anyways it is not encouraged to do cancellations in the right half plane or outside the unit disk), so I will pay attention to the poles directly. And I will calculate `G(s=0)` and `G(z=1)` in the limit. As a result, the user needs to call the `minreal` or some other function to explicitly demand pole-zero cancellations. The output of `minreal` will be of transfer function and/or state-space form, and hence, the `dcgain` function should work fine.

We can talk more about it when I create the pull request.

closed 2023-01-05 by baggepinnen