JuliaControl/ControlSystems.jl #17: bode for SISO returns all data in last dim

mfalt, 2016-03-11:
_From @baggepinnen on February 22, 2016 19:25_

This behavior causes lots of squeeze statements
```julia
julia> Control.bode(S)
(
1x1x200 Array{Float64,3}:
[:, :, 1] =
 1.00504

[:, :, 2] =
 1.04091

[:, :, 3] =
 1.07809

...
```

_Copied from original issue: mfalt/Control.jl#22_

mfalt, 2016-03-11:
_From @baggepinnen on February 22, 2016 19:26_

The same goes for `freqresp` and probably more funs

mfalt, 2016-03-23:
This should be a priority (and will be breaking some existing code), but should be included in one of the next versions.

baggepinnen, 2016-03-23:
What if bode returned an nu x ny matrix with vectors? This would align well
with a TF being an nu x ny matrix of siso systems.
In the siso case, the vector could be automatically popped out of the 1 by
1 matrix

mfalt, 2016-03-23:
But the problem then is that we have to have a special case for siso systems, I think the solution above is more elegant and user friendly.

mfalt, 2016-05-03:
Fixed in a2c7bbf

closed 2016-05-03 by mfalt