JuliaControl/ControlSystems.jl #25: Name change of package

mfalt, 2016-03-14:
To be able to put this toolbox in julia METADATA it seems like we need to change name to ControlSystems.jl instead of Control.jl, does anyone have any objections or thoughts about this? 
@jcrist?
I would prefer Control.jl but it seems unlikely that they would accept this (see https://github.com/JuliaLang/METADATA.jl/pull/4753 for discussion). "ControlSystem" is the name that matlab uses for example, is this better than "ControlSystems"?

baggepinnen, 2016-03-14:
+1 for ControlSystems.jl. It's a well established name (except for the plurality) due to the Matlab toolbox which has been widely used in academia for a long time. I also think that ControlSystems is better than ControlSystem since many julia packages are named with the plural form of a noun.

phelipe, 2016-03-15:
The julia documentation says:
"Packages that provide most of their functionality in association with a new type should have pluralized names.

    DataFrames provides the DataFrame type.
    BloomFilters provides the BloomFilter type.
    In contrast, JuliaParser provides no new type, but instead new functionality in the JuliaParser.parse() function.
"

jleny, 2016-03-15:
+1 for ControlSystems.jl
The singular in Matlab I think only applies because it is followed by "Toolbox".

This package is starting to look quite good, by the way. Thanks for the effort.

mfalt, 2016-03-15:
Thanks for all your input, we settled for ControlSystems.jl so I will close this now. Although we do not provide a type called `ControlSystem`, the plurality is still common among julia packages and should not cause any confusion.

closed 2016-03-15 by mfalt