JuliaControl/ControlSystems.jl #28: How to handle sample times

mfalt, 2016-03-17:
It is currently possible to do
```julia
f = tf(1,[1,1])
[f 0; 0 f]
```
without any problems but the discrete version
```julia
f = tf(1,[1,1], 0.01)
[f 0; 0 f]
```
throws an error that sampling times are not equal. This is because
```julia
Base.convert{T<:Real}(::Type{TransferFunction}, b::T) = tf([b], [1])
```
which is called through `promote` in `(h)vcat` defaults to continuous time (`Ts=0`).
How should we handle this? I can see two solutions:
###### 1. Keep the code as it is.
If the user wants to create a matrix like this it is possible to do
```julia
d0 = tf(0, 0.01)
f = tf(1,[1,1], 0.01)
[f d0; d0 f]
```
The downside is that we are treating continuous and discrete systems differently without any real reason.
###### 2. Default all transfer-functions, that are not explicitly created, to a sampling time of e.g. `Ts=-1`.
It is then possible to identify this while concatenating and convering to the appropriate value, and we can throw errors if the user is trying to for example simulate a system like this.

jleny, 2016-03-19:
I would interpret [f 0;0 f] as a transfer matrix with sampling time 0.01 in your example. It seems to me that the promotion rules for transfer functions should be revised. Currently
f, g = promote(f,0) returns a discrete-time tf for f, and a continuous-time tf for g. That does not seem natural.

mfalt, 2018-09-10:
The current plan is to move Discrete/Continuous into the type system together with a type for "Static" systems. Using this issue to track that.

baggepinnen, 2020-10-10:
Closed by #315

closed 2020-10-10 by baggepinnen