JuliaControl/ControlSystems.jl #48: Roadmap

ufechner7, 2016-03-29:
What is the problem, that this package shall solve? Which functionality shall be implemented?
a) linear system design
b) PID controller design
c) optimal control design
d) non-linear control system design
Shall it be used for a specific type of control system course (teaching)? 
If yes, which level of courses?
Shall all the functions of Scilab be implemented?
See: https://help.scilab.org/docs/5.5.2/en_US/section_64a8529216e858b335b0e6c058385350.html

Any comments welcome.

baggepinnen, 2016-03-29:
I think there will be a few nice LQG examples available within a few weeks. Examples serve as great demonstrators as to what the toolbox is capable of, besides being of great help when one is starting out using the toolbox.

I think the toolbox should handle everything we use in the basic control course at our department, except maybe for the simulink related stuff, which is harder to replace. Communication with processes and lab equipment is another thing I think is outside the scope of this toolbox, but nice interfaces to eventual hardware packages would of course be interesting. If anyone know of suitable such hardware packages, it might be a good idea to have a look at them early on and at least keep their functionality in mind.

I also have some loose plans on implementing some adaptive control related tools in the toolbox. The design of controllers using simple adaptive schemes like self-tuning regulators, RLS/kalman parameter estimators, MIT-rule and other gradient based methods would be nice to have support for, not only for synthesis, but also during simple simulations with user selectable reference trajectories and noise.

ma-laforge, 2016-12-27:
@baggepinnen:
> Communication with processes and lab equipment is another thing I think is outside the scope of this toolbox, but nice interfaces to eventual hardware packages would of course be interesting. If [...]

Assuming your field uses HPIB/GPIB/LXI equipment as well,

 1. I have used the PyVISA module in the past:
https://pyvisa.readthedocs.io/en/stable/

but that will require a wrapper module for Julia.

2. There is the "Instruments.jl" module registered with METADATA.  It appears to be developed by Raytheon BBN Technologies.  I myself have not used this module.

3. There is also "NIDAQ.jl" module: "National Instruments' driver for their data acquisition boards."  This one appears to be developed by the Howard Hughes Medical Institute.  I have never required/used anything like this - so I cannot comment here either.

Hope this helps.

closed 2023-01-05 by baggepinnen