JuliaControl/ControlSystems.jl #59: "norm" for stable system returns "Inf"

denglerchr, 2016-05-17:
`sys = tf([1, -1.45, 0.475], [1, -2*0.5, 0.5^2], 1.0); 
norm(sys)`
returns "Inf". Both poles are at 0.5, so the norm should be bounded. Same code in Matlab returns 1.1269.

mfalt, 2016-05-17:
Thanks. I fixed this in pull request #60, it was a bug with the direct term causing it to be Inf even if the system is discrete. The new version returns `1.1269427669584644`. I will pull this as soon as possible, but it may take a few weeks before we tag a new version.

mfalt, 2016-05-22:
I pulled this into master now, to get the changes you have to do a
`Pkg.checkout("ControlSystems")`
or wait until we tag a new version and do a
`Pkg.update()`
Thanks for reporting this.

closed 2016-05-22 by mfalt