JuliaControl/ControlSystems.jl #63: A better covar

mfalt, 2016-05-17:
A brief repetition of what covar is supposed to do for continuous systems:

> Calculate the stationary covariance P of the system output y(t) when the input has the property E[u(t)u(tau)'] = W*delta(t-tau), where delta(t) is the dirac delta.

The current implementation of covar returns a covariance matrix of Inf if the system is unstable or if there is a direct term` D .!= 0.` This is quite reasonable and the same as current versions of matlab. It could however be done better. The current algorithm does the following
```
Q = lyap(A, B*W*B')
P = C*Q*C' + D*W*D'
```
with the extra case of when ` D .!= 0` and system is unstable. However, some of the outputs might still have definable variance on the output (if for example the noise is correlated so that it doesn't pass through the direct term `D*W*D' = 0`. A solution for this would be
```
Q = lyap(A, B*W*B')
P = C*Q*C'
P[D*W*D' .!= 0] = Inf
```
But we also have the case where some states are not controllable/observable. It would be nice to be able to capture these cases properly. For example `A=1, B=C=D=0` would generate a zero output, but we would output Inf. And `A=B=C=D=0` throws an error because `lyap` is not solvable (or has no unique solution).

baggepinnen, 2019-01-30:
Seems addressed in #61 
Can we close @mfalt ?

mfalt, 2019-01-30:
Seems like it

closed 2019-01-30 by mfalt