JuliaControl/ControlSystems.jl #69: Time response for discrete system with specified final time throws error

olof3, 2016-06-29:
```julia
G = tf([1], [1; zeros(3)], 1)
y, t, x = step(G, 10)
```
gives

ERROR: Time vector must match sample time for discrete system

The problem seems to be that the wrong time vector is created for the step response.

mfalt, 2016-06-29:
Nice catch, it was indeed the default time vector that was one step too short so it became
`0, 0.1111, 0.222, ...., 10`
instead of
`0, 0.1, 0.2, ..., 10`
I fixed this in fd896a1, on branch `defaultTimeFix`. We have some problems with the plots so I don't want to pull it just yet, if you want to use the version with the fix do
`Pkg.checkout("ControlSystems", "defaultTimeFix")`
I will close once it is pulled.

olof3, 2016-06-29:
Maybe more intuitive to use 
```julia
0:Ts:Tf
```

than

```julia
linspace(0, Tf, round(Int, Tf/Ts)+1)
```
the second option doesn't work if the final time isn't a multiple of the sample time.

Can I add a pull requrest to the branch you created? In that case I would be happy to add a test case, and possibly change the returned time vector.

mfalt, 2016-06-29:
You are right, I am not sure why it was written like that, maybe to guarantee that Tf will be the last sample. Maybe to ensure that the simulation goes to at least Tf, we should have
`0:Ts:(ceil(Tf/Ts)*Ts)`
Feel free to add a pull request.

commit 0a0622f1 (2016-07-01): Fixes #69 - Fix to avoid problem when final time is not multiple of sample time. Added tests. (#73)
PR #74 (2016-07-04): Default time fix
Fixes #69
closed 2016-07-04 by mfalt