JuliaControl/ControlSystems.jl #72: Update plot functions to recipes?

baggepinnen, 2016-06-29:
I create this issue to facilitate a discussion on whether to update all or a subset of the plot functions to recipes
http://plots.readthedocs.io/en/latest/recipes/
I have updated a few other packages to recipes and will create a branch where I translate a single function, like `bodeplot` to a recipe to see how it works out.
Some pros and cons I have noted so far

Pros:
- Become very flexible in creating plots.
- Plot functions automatically accept plotting related keyword args.
- Creation of complicated plots and layouts becomes easier

Cons:
- Requires significant rewriting of code
- Recipes are **much** harder to debug due to errors being thrown from code written in a macro, there will thus be no line number info, not even any help as to what error was thrown.

mfalt, 2016-06-29:
Great initiative.
Could you give an example of how the syntax for creating a bode plot (for the user) might look?

baggepinnen, 2016-06-29:
The syntax for the user would not change at all (I think). The ability to create complicated composite plots would be new to the user though, and I'll try to come up with a meaningful example when I have a draft of one or a few recipes.

baggepinnen, 2017-08-29:
Closed with #103

closed 2017-08-29 by baggepinnen