JuliaControl/ControlSystems.jl #75: stepplot not working

phelipe, 2016-07-17:
Is step plot working ? 

LoadError: UndefVarError: subplot not defined
while loading In[6], in expression starting on line 10

 in stepplot at /home/phelipe/.julia/v0.4/ControlSystems/src/plotting.jl:81
 in stepplot at /home/phelipe/.julia/v0.4/ControlSystems/src/plotting.jl:103

mfalt, 2016-07-18:
Which version of ControlSystems are you using?

We had a problem when the syntax for subplots where changed in Plots.jl which caused the problem you describe, and fixed it in version 0.1.4. It has however not been accepted in the official package list because we have to change a feature in our tests. You can get the working version by running `Pkg.checkout("ControlSystems")

 I opened a separate issue for this, with some explanation, here: #76

phelipe, 2016-07-19:
I was using Pkg.update() to upgrade the packages but only Pkg.checkout("ControlSystems") solved the problem to Pyplot backend.  Using Gadfly I get an warning: 

"WARNING: Backend gadfly has been deprecated.  It may not work as originally intended.".

Anyway the problem was solved.
Thank you!

closed 2016-07-19 by phelipe