JuliaControl/ControlSystems.jl #76: Restructure tests for plotting

mfalt, 2016-07-18:
We are not allowed to checkout unregistered packages in the tests so have to change how we test the plots. The current structure is using the ControlExamplePlots.jl, we should restructure it so that we download specific versions of that repo instead of using `Pkg.checkout`. See discussion on pull request [here](https://github.com/JuliaLang/METADATA.jl/pull/5545) and on Plots.jl [here](https://github.com/tbreloff/VisualRegressionTests.jl/issues/5)

This is urgent because of for example #75.

mfalt, 2016-07-28:
I fixed this now and we should be allowed to register the new version. We should however consider checking out specific versions of the test images. However, when the version of `Plots.jl` changes, the test will likely fail again, making it hard to verify old versions. We should probably wait with a long term solution until `VisualRegressionTests.jl` have converged on a solution.

baggepinnen, 2020-10-10:
This issue seem stale, recent discussion in #353

closed 2020-10-10 by baggepinnen