JuliaControl/ControlSystems.jl #81: Poles for MIMO system

olof3, 2016-08-04:
The poles of a transfer matrix is not the concatenation of the poles of the matrix elements.

Compare for example Example 3.1 in Control Theory by Glad&Ljung.
which concludes that
```
s=tf("s"); sort(pole([2/(s+1) 3/(s+2); 1/(s+1) 1/(s+1)]))  == [-2, -1, -1]`
`'`

another test case is
```
s=tf("s"); pole(tf(ss([-2], ones(1,2), ones(2,1), zeros(2,2)))) == [-2]
```

I don't know how it should be handled in a numerically sound way.

olof3, 2016-08-04:
The test case
```
@test_approx_eq pole([sys sys]) [-1.0, 4.0, -4.0, -1.0, 4.0, -4.0]
`'`
is actually not correct.

mfalt, 2019-01-30:
https://github.com/JuliaControl/ControlSystems.jl/pull/165

closed 2019-01-30 by mfalt