JuliaControl/ControlSystems.jl #97: Problem with graphics

cjruiz, 2017-05-16:
Hello,
I have a non-sense problem. When i use Step, or somtehing like that, i never get a graphic.
For example: 
`using ControlSystems
g = tf(1,1)
step (g)
`

The output is: ([1.0; 1.0; . ; 1.0; 1.0],
0.0:1.0:100.0,)

I cant figure whats going on. Any idea?

baggepinnen, 2017-05-16:
Hello,

Try the function `stepplot` instead. All Plots are produced by functions
that end with - plot, as bodeplot or nicholsplot.

For this to work you need the packages Plots.jl and a suitable backend,
such as Pyplot or GR.jl

Sent from mobile device

cjruiz, 2017-05-16:
Hi,
first of all, thanks for the answer.

Unfortnately, it didn't work. Now i have a new error: 
Key error: key "set_facecolor" not found

I have Plots.jl and PyPlot.jl- Actually, it was working a few hours ago, but now i get nothing, and i can't understand. I have no problems plotting arrays, only with ControlSystems. 

Sorry and thanks again.

mfalt, 2017-05-16:
It is hard to debug this without a stack-trace, could you post the full error you are getting? Which version of julia are you running? Have you updated Plots.jl and PyPlot.jl?

cjruiz, 2017-05-16:
Julia version -> 0.5.0

Yes, i updated:
Plots version ->0.11.2
PyPlot version ->2.3.2

cjruiz, 2017-05-16:
Well, is working now. I don't know why, but now it works. After the number 10000 call to bodeplot() Julia started to recompiling and stalling new packages.
[image]

closed 2017-05-16 by cjruiz