JuliaControl/ControlSystems.jl #101: How can i create a tf with delay?

cjruiz, 2017-08-09:
I want to create a tf with delay. It is possible? How?

Thanls

mfalt, 2017-08-09:
Unfortunately we don't have that capability yet. It would require an alternative representation of transferfunctions. This is part of the TODO in the restructuring, see, e.g., https://github.com/JuliaControl/ControlSystems.jl/issues/68#issuecomment-225563153 but no-one is actively working on it as far as I know.

cjruiz, 2017-08-09:
Ok, i see, thanks!

Can i use this issue to ask for an example of a MIMO system, also using kwargs? I'm trying to create one but i don't get it. Thanks

baggepinnen, 2017-08-09:
Please ask your question in the gitter channel at https://gitter.im/JuliaControl/public?source=orgpage

closed 2017-08-09 by cjruiz