JuliaControl/ControlSystems.jl #129: bodeplot - 'latex' is not recognized as an internal or external command, operable program or batch file.

dlwalter, 2018-05-16:
I'm getting an error when using bodeplot.  The following code causes it:

```
using ControlSystems
s = tf("s")
H = 1/(s+1)
bodeplot(H, w)
```

I get the following message:

```
'latex' is not recognized as an internal or external command,
operable program or batch file.
'latex' is not recognized as an internal or external command,
operable program or batch file.
latex: failed to create a dvi file
latex: failed to create a dvi file
```

It also creates a tex file in the directory named:
33865466217e273735d668289c47cd88.tex

baggepinnen, 2018-05-17:
That's interesting.. which backend do you have activated for `Plots.jl`? Try if possible change backend to see if it's backend specific.

baggepinnen, 2018-05-18:
Do you see the same issue if you call plots yourself?
```julia
using Plots
plot(randn(10))
gr()
plot(randn(10))
```

dlwalter, 2018-05-18:
Ran you code no problem.  Plot shows up and no messages.  I guess that it
may be because this plots code doesn't run any latex text.

When bodeplot generates those errors I also notice that the bodeplot legend
is blank.

dlwalter, 2018-05-18:
[image]

zhiyuanryanchen, 2020-04-03:
[image]

I have a similar problem using Julia 1.4.0

baggepinnen, 2020-04-03:
The gr backend has suffered from a lot font issues,. You can change backend if you'd like a nicer looking output, I prefer plotly for almost everything nowadays.

baggepinnen, 2020-10-10:
Does not appear to be a problem anymore :)

closed 2020-10-10 by baggepinnen