JuliaControl/ControlSystems.jl #166: Multiplying 1x1 tf with mxn should result in elementwise multiplication?

albheim, 2018-11-13:
I think this would make sense since that is how it works in matlab. Could also be done with julia's broadcasting, but to me it seems smoother to just check if it's 1x1 and then do element-wise multiplication.

`a = [s 1]

b = s

a*b # Right now this produces an error since a is 1x2 and b is 1x1`

mfalt, 2018-11-14:
Although I don't buy argument "I think this would make sense since that is how it works in matlab.", there is something here that can be improved.

It relates closely to the discussion of encoding SISO/MIMO in the type parameters. (This has several other advantages where we can drop indices and dispatch for siso)

The system 'b'/'s' would then by default be a siso system for which we might want to automatically broadcast, while still enforcing the dimensions of mimo systems including 1x1 "mimo" to agree.

On the other hand, I think 'a.*b' is a reasonable syntax for the desired behavior too.

olof3, 2018-11-15:
I think this syntax would be nice for something like constructing MIMO transfer function matrices like

`1/(s+1)*[1 s; 1+s 1]`

but then again, I guess it really would make more sense to use `.*` instead.

I think it is would be better to not allow multiplication of SISO and MIMO systems with `*` since it would make it harder to catch issues with messed up dimensions. Multiplication of MIMO system with  UniformScaling{SISO system} would perhaps be an alternative if there is a need.

mfalt, 2018-11-16:
Although
```julia
a = 1/(s+1) # SISO
b = [1 s; 1+s 1] # MIMO
c = a*b
```
could throw an error to catch unintended dimension mismatch, I think
SISO * MIMO meaning element-wise is pretty natural. Just like for scalars and matrices.
However, with UniformScaling, I believe there is only one right choice
```julia
a*I # SISO
b*I # MIMO
a*I+b ==  [1/(s+1) 0; 0  1/(s+1)]*b
```
since `I` is the "multiplicative identity".

olof3, 2019-01-24:
The direct interpretation of SISO * MIMO amounts to the connectiong of many outputs to a single input.

As suggested by @mfalt in an offline discussion, it is certainly possibly to define  SISO * MIMO, and have some sort of interpretation of what this means. I.e. filtering all outputs of the MIMO system through the same filter.

In cases where one has made a mistake it would be a great help if  SISO * MIMO failed, and in those cases where it is intentional, a tiny amount of extra notation is not a big deal.

olof3, 2019-01-24:
Along the same line of reasoning  it seems dubious to allow addition of scalars and MIMO systems, i.e.
```
ss(-ones(2,2), ones(2,2), ones(2,2), zeros(2,2)) + 1
```

albheim, 2020-09-24:
Found this old issue and thought it might be time to close it?

I agree with what is said here, and think it works well as it is. The one thing is maybe olofs previous point, questioning if addition of scalar and MIMO should be allowed normally? It seems to work like a broadcasting operation right now, which makes me wonder why it is not implemented as such?

closed 2020-12-21 by albheim