JuliaControl/ControlSystems.jl #194: lsim breaks down for scalar u or x0

hurak, 2019-03-10:
`lsim` fails when the control signal `u` is just a scalar and is entered as such:
```
using ControlSystems

a = b = c = 1
G = ss(a,b,c,0)
u(x,t) = 0.0
t = 0:0.1:1
y, t, x, uout = lsim(G,u,t)
```

> ERROR: LoadError: return value of u must be of size nu
Stacktrace:
 [1] #lsim#112(::Array{Float64,1}, ::Symbol, ::Function, ::StateSpace{Int64,Array{Int64,2}}, ::typeof(u), ::StepRangeLen{Float64,Base.TwicePrecision{Float64},Base.TwicePrecision{Float64}}) at /home/hurak/.julia/packages/ControlSystems/6Nac2/src/timeresp.jl:149
 [2] (::getfield(ControlSystems, Symbol("#kw##lsim")))(::NamedTuple{(:x0,),Tuple{Array{Float64,1}}}, ::typeof(lsim), ::StateSpace{Int64,Array{Int64,2}}, ::Function, ::StepRangeLen{Float64,Base.TwicePrecision{Float64},Base.TwicePrecision{Float64}}) at ./none:0
 [3] top-level scope at none:0
 [4] include at ./boot.jl:326 [inlined]
 [5] include_relative(::Module, ::String) at ./loading.jl:1038
 [6] include(::Module, ::String) at ./sysimg.jl:29
 [7] include(::String) at ./client.jl:403
 [8] top-level scope at none:0
in expression starting at /home/hurak/ownCloud/julia/lsim_breaking_for_siso.jl:8

As soon as both `u` is modified to 
```
u(x,t) = [0.0]
```
things are fine.

The same problems for the initial state `x0` in
```
using ControlSystems

a = b = c = 1
G = ss(a,b,c,0)
u(x,t) = [0.0]
t = 0:0.1:1
x0 = 1.0
y, t, x, uout = lsim(G,u,t,x0=x0)
```

> ERROR: LoadError: TypeError: in keyword argument x0, expected Union{Array{T,1}, Array{T,2}} where T, got Float64
Stacktrace:
 [1] (::getfield(ControlSystems, Symbol("#kw##lsim")))(::NamedTuple{(:x0,),Tuple{Float64}}, ::typeof(lsim), ::StateSpace{Int64,Array{Int64,2}}, ::Function, ::StepRangeLen{Float64,Base.TwicePrecision{Float64},Base.TwicePrecision{Float64}}) at ./none:0
 [2] top-level scope at none:0
 [3] include at ./boot.jl:326 [inlined]
 [4] include_relative(::Module, ::String) at ./loading.jl:1038
 [5] include(::Module, ::String) at ./sysimg.jl:29
 [6] include(::String) at ./client.jl:403
 [7] top-level scope at none:0
in expression starting at /home/hurak/ownCloud/julia/lsim_breaking_for_siso.jl:8

In my opinion this might be confusing, especially to newcomers.

mfalt, 2019-03-11:
I agree that it might be confusing, and we should make sure to have a consistent interface to `lsim`, which I'm not sure we have.
One solution to this might be something similar to what we are doing in the delay_lti branch (for simulating systems with internal delays). There we have a wrapper automatically converting numbers, and enforcing in-place computations internally.
See 
https://github.com/JuliaControl/ControlSystems.jl/blob/baac22dab3fbced19b9d7708e6ecadc094d1f82b/src/delay_systems.jl#L50
However, for sytsems with multiple inputs, this could create unexpected results for the user?

mfalt, 2019-08-29:
You are completely right, the docs should be updated to reflect that u has to be a vector. I think we initially allowed scalar u when the update was done using broadcast, but this could easily result in hidden bugs when there are multiple states and doesn't play nice when DifferentialEquations is used to simulate the system, that's why we require a vector.

baggepinnen, 2020-10-10:
Fixed in #319

closed 2020-10-10 by baggepinnen