JuliaControl/ControlSystems.jl #195: lsim not working for systems that are not strictly proper

hurak, 2019-03-10:
Apparently, `lsim` does not work for SISO systems that are not strictly proper, that is, it fails for systems whose `D` term in the state space description is nonzero. I have not checked the case in which `D` is a matrix. 
```
using ControlSystems

a = b = c = d = 1
G = ss(a,b,c,d)
u(x,t) = [0.0]
t = 0:0.1:1
x0 = [1]
y, t, x, uout = lsim(G,u,t,x0=x0)
```

> ERROR: LoadError: AssertionError: Can not simulate systems with direct term D != 0
Stacktrace:
 [1] Simulator(::StateSpace{Int64,Array{Int64,2}}, ::typeof(u)) at /home/hurak/.julia/packages/ControlSystems/6Nac2/src/simulators.jl:41
 [2] #lsim#112(::Array{Int64,1}, ::Symbol, ::Function, ::StateSpace{Int64,Array{Int64,2}}, ::typeof(u), ::StepRangeLen{Float64,Base.TwicePrecision{Float64},Base.TwicePrecision{Float64}}) at /home/hurak/.julia/packages/ControlSystems/6Nac2/src/timeresp.jl:165
 [3] (::getfield(ControlSystems, Symbol("#kw##lsim")))(::NamedTuple{(:x0,),Tuple{Array{Int64,1}}}, ::typeof(lsim), ::StateSpace{Int64,Array{Int64,2}}, ::Function, ::StepRangeLen{Float64,Base.TwicePrecision{Float64},Base.TwicePrecision{Float64}}) at ./none:0
 [4] top-level scope at none:0
in expression starting at /home/hurak/ownCloud/julia/lsim_breaking_for_siso.jl:8

Any plans to make it working for systems for nonzero `D` term in the state space description? Were there any issues?

mfalt, 2019-03-11:
This is unfortunately a result of us switching to a differential equations solver for simulating continuous systems. If zero-order-hold is used to simulate a system, this is not a problem, so a workaround is simply using
` y, t, x, uout = lsim(G,u,t,x0=x0, method=:zoh)`
However, I see no reason why we should not be able to add the direct term. It may be as easy as to uncomment the direct term in https://github.com/JuliaControl/ControlSystems.jl/blob/master/src/simulators.jl#L39
Feel free to investigate if this solves the problem and if the tests still pass.

baggepinnen, 2019-03-11:
The commen was introduced in https://github.com/JuliaControl/ControlSystems.jl/commit/e0bb7c51475cc37a95a29fc4bc5d2d5da044df9d due to stack overflow

closed 2023-04-25 by baggepinnen