JuliaControl/ControlSystems.jl #197: Hot to get lsim working for a discrete-time system?

hurak, 2019-03-11:
I have difficulties making the lsim function run for a discrete-time system. See the code below:
```
using ControlSystems

a = b = c = 1.0
d = 0.0
G = ss(a,b,c,d)

Ts = 1
Gd = c2d(G,Ts)

t=0.0:Ts:10

u(x,t) = 0
#u = zeros(11)

y, t, x, uout = lsim(Gd,u0,t)
```

Neither setting `u` to a function returning zero nor setting it to a zero sequence (vector) works.  

In the documentation for the lsim function I can read 

> Continuous time systems are simulated using an ODE solver if u is a function. If u is an array, the system is discretized before simulation. For a lower level inteface, see ?Simulator and ?solve
>
>u can be a function or a matrix/vector of precalculated control signals. If u is a function, then u(x,i) (u(x,t)) is called to calculate the control signal every iteration (time instance used by solver). This can be used to provide a control law such as state feedback u(x,t) = -L*x calculated by lqr. To simulate a unit step, use (x,i)-> 1, for a ramp, use (x,i)-> i*h, for a step at t=5, use (x,i)-> (i*h >= 5) etc.

Based on the first paragraph, does it mean that it is not possible to simulate a discrete-time system with `u` given as a function of `x`, that is, a state feedback? It would be a pity, would not it? It appears the text is saying that for a discrete-time system it is only possible to simulate its response to some precomputed sequence. And even that I fail to do.

baggepinnen, 2019-03-11:
Gd in your example is a tuple, try using Gd[1]

mfalt, 2019-03-11:
To clairify, as the documentation (i.e`?c2d`) says:

```julia
"""[sysd, x0map] = c2d(sys, Ts, method=:zoh)
...
Returns the discrete system `sysd`, and a matrix `x0map` that transforms the
initial conditions to the discrete domain by
x0_discrete = x0map*[x0; u0]"""
```

hurak, 2019-03-11:
> To clairify, as the documentation (i.e`?c2d`) says:
> 
> ```julia
> """[sysd, x0map] = c2d(sys, Ts, method=:zoh)
> ...
> Returns the discrete system `sysd`, and a matrix `x0map` that transforms the
> initial conditions to the discrete domain by
> x0_discrete = x0map*[x0; u0]"""
> ```

Thanks for the clarification. The syntax seems so seducingly similar to what we all know from Matlab that I was not able to notice this tiny (yet important, of course) difference while reading the help. Thanks.

closed 2019-03-12 by mfalt