JuliaControl/ControlSystems.jl #199: iscontinuous function not found

hurak, 2019-03-12:
Why is't the `iscontinuous` function among the exported (and thus available to users)? I am sure the function is very useful for users, why not exporting it? Was it just an unintended omission?

olof3, 2019-12-19:
Yes, seems like it would be a good idea to export a function with that functionality.

To me `iscontinuous` feels a little bit imprecise though, the state of a discrete-time linear system is also continuous, and one could think that `iscontinuous` is true for these systems while being false for boolean statespace models. 

On the other hand, function names like `iscontinuoustime` or  `is_continuous_time` are quite long.
But since the function would probably be used only once in each call, and since it should "soon" be possible to dispatch on continuous and discrete time, it seems like `is_continuous_time` is a reasonable function name given that it is more precise. I would say I have a weak preference for  `is_continuous_time`. But perhaps someone has a better suggestion altogether.

Anyway, the corresponding `isdiscrete` or `is_discrete_time` should also be exported as well.

hurak, 2019-12-19:
I agree that `iscontinuous` is imprecise. And I like the proposed `is_continuous_time` and `is_discrete_time`.

freemin7, 2020-02-17:
Idea:
iscontinous means continous both in value and in time
iscontinous can take an additional type to care just about being continuous in value or being continuous in time.

mfalt, 2020-04-07:
> iscontinous means continous both in value and in time

I am not sure what this would mean in practice. I don't think we have any systems that are not continuous in "value".

I agree that `iscontinuous` is slightly ambiguous, but I am not a big fan of `is_continuous_time`. Whatever we choose, it will be exported once we pull this https://github.com/JuliaControl/ControlSystems.jl/pull/256 gigantic PR, so we have to make a decision.

baggepinnen, 2020-04-08:
I guess an integer statespace model could be considered as having a discrete state space?

olof3, 2020-04-08:
Is it the length (6 extra characters) that is problem? The function will would not be used that often, at most once in a few functions here and there, and even less when it will be possible to dispatch on the sample time. To me, the slightly longer name is outweighed by the improved clarity.

closed 2023-04-25 by baggepinnen