JuliaControl/ControlSystems.jl #200: obsv fails when the two matrices are of different type

hurak, 2019-03-13:
```
A = [1 2; 3 4]
B = [5.6; 7.8]
obsv(A,B)
```

> ERROR: MethodError: no method matching obsv(::Array{Int64,2}, ::Array{Float64,1})
Closest candidates are:
  obsv(::AbstractArray{T<:Number,2}, ::AbstractArray{T<:Number,2}) where T<:Number at /home/hurak/.julia/packages/ControlSystems/RDC5Q/src/matrix_comps.jl:111
Stacktrace:
 [1] top-level scope at none:0

hurak, 2019-03-13:
> Well this is not a bug. First of all, your code could never work since `obsv` expects the `C` matrix, and your matrix `B` has the wrong dimensions to act as such. 

Indeed, I mixed up a few thing while copy&pasting and polishing them here (the name `B` suggests where it comes from). Sorry, it's those late night shifts... Thanks for the patience.

>Second, the function can not know what return type you want if the matrices are not of the same type

I understand.

mfalt, 2019-03-13:
> Second, the function can not know what return type you want if the matrices are not of the same type, consider the following
I don't think I agree with this. The return type can simply be inferred by the promote type of elements in A and C, in this case Float64.
I think that we in general are too restrictive in our typing of arguments in our functions.

baggepinnen, 2019-04-16:
closed with #205

closed 2019-04-16 by baggepinnen