JuliaControl/ControlSystems.jl #208: Register

baggepinnen, 2019-04-18:
@JuliaRegistrator register()

JuliaRegistrator, 2019-04-18:
Registration pull request created: [JuliaRegistries/General/122](https://github.com/JuliaRegistries/General/pull/122)

After the above pull request is merged, it is recommended that you create
a tag on this repository for the registered package version:
```
git tag -a v0.5.2 -m "<description of version>" 7848c13dc86ce6f8a21ca589162ff1907c9820bb
git push origin v0.5.2
```

baggepinnen, 2019-05-22:
@JuliaRegistrator register()

closed 2019-05-22 by baggepinnen

JuliaRegistrator, 2019-05-22:
Error while trying to register: Version 0.5.2 already exists

JuliaRegistrator, 2019-05-22:
Registration pull request created: [JuliaRegistries/General/910](https://github.com/JuliaRegistries/General/pull/910)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.5.3 -m "<description of version>" d300320d5496928a86eae21491622960f208e447
git push origin v0.5.3
```

JuliaRegistrator, 2020-03-31:
Registration pull request created: [JuliaRegistries/General/11854](https://github.com/JuliaRegistries/General/pull/11854)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.5.8 -m "<description of version>" c5efbcb60774df84c547c960454b495df9bcf1b1
git push origin v0.5.8
```