JuliaControl/ControlSystems.jl #213: The status of balreal, baltrunc and other functions related to model order reduction

hurak, 2019-05-21:
I have noticed that there are some functions related to model order reduction in [matrix_comps.jl](https://github.com/JuliaControl/ControlSystems.jl/blob/master/src/matrix_comps.jl) but they are not listed anywhere. Namely `balreal()` and `baltrunc()` I only found them while going through the code. Is there any reason for hiding them (perhaps users are not encouraged to use them right now)? If not, it would be a pitty not to list them. I might even contribute some examples of usage.

baggepinnen, 2019-05-21:
Both `balreal` and `baltrunc` are exported names, but they might not have ended up in the documentation which they of course should do. They should be perfectly usable, at least last time I checked..

baggepinnen, 2019-05-21:
See #212 . They might get included if that PR is merged

baggepinnen, 2022-01-04:
This can safely be closed now, they appear in the docs here
https://juliacontrol.github.io/ControlSystems.jl/latest/lib/analysis/#ControlSystems.baltrunc-Tuple{ST}%20where%20ST%3C:AbstractStateSpace
and there are two PRs up making them a whole lot better
- #588
- #589

closed 2022-01-04 by baggepinnen