JuliaControl/ControlSystems.jl #228: Equivalent to Simulink "Saturation" block

non-Jedi, 2019-08-28:
When simulating real-world systems, the saturation block in simulink is very important. Would it be possible to include something like that to compose with transfer functions?

baggepinnen, 2019-08-28:
The saturation is a nonlinear function, whereas transfer functions are linear. You could simulate a saturation on the input using the function interface to lsim or the Simulator type, but for more advanced usage you would have to resort to Differentialequations.jl directly.

baggepinnen, 2019-08-28:
No, connecting systems like that is not something that is supported so far. You may either try your luck with Differentialequations.jl directly, or possibly one of the modeling languages like Modia.jl or Modelingtoolkit.jl

mfalt, 2019-10-29:
This is definetely something where we could consider creating a simple API for creating the required differential equations (in the case of no direct terms). For example
```julia
dGp, fGp = diffeq(Gp)
dGc, fGc = diffeq(Gc)
# Functions so that dG returns derivative of internal state and
# u = fGc(x1), where x1 is internal state of controller
# y = fGp(x2) , where x2 is internal state of process

# Some definiton of feedback that could be returned by Controllsystems.jl
function dx(x,r)
    x1 = # appropriate index in x vector
    x2 = # appropriate index in x vector
    u = fGc(x1)
    y = fGp(x2)
    e = r - y
    [dGc(x1, e); dGp(x2, u)]
 end
y(x) = fGp(x["indices for x2"])

```
The `dx` function can now be used in DifferentialEquations.jl, and `y(x)` would return output.

There are some details regarding efficiency and direct terms, but in general it would make it easier for the user to add nonlinearities and build their own differential equations.

baggepinnen, 2019-10-30:
The [`Simulator`](https://github.com/JuliaControl/ControlSystems.jl/blob/9f3bd5d5ab7b275ed8443e6ab876ac2b12077fd1/src/simulators.jl#L42) type already does more or less this. A library of nonlinearities feels somewhat out of scope, because people would then want not only saturations, but all other common nonlinearities as well. It would perhaps fit well into an add-on package, but making use of the existing ecosystem for simulation is probably a more powerful apporach.

mfalt, 2019-10-30:
It is not clear how you would use the Simulator directly for the feedback case. I agree that we don't need a library of non-linearities, but what we could add is some simple manipulations (for example on the Simulator type) such as feedback with other Simulators.

baggepinnen, 2021-03-11:
ComponentArrays.jl was built in part to facilitate composing dynamical systems together, they have a tutorial using ControlSystems here
https://jonniedie.github.io/ComponentArrays.jl/dev/examples/adaptive_control/#Helper-Functions-1
The interface in that tutorial is rather low-level, but I believe it could be made very approachable with a slightly higher-level interface built on top.

baggepinnen, 2022-05-16:
https://juliacontrol.github.io/ControlSystems.jl/latest/lib/nonlinear/#Control-signal-saturation