JuliaControl/ControlSystems.jl #229: showterm2 function out of date

andrersimoes, 2019-10-21:
Hi gentlemen,

I'm receiving an error inside the **showterm2** (polyprint.jl) function regarding the number of parameters used at **Polynomials.printsign** call.

``
ERROR: MethodError: no method matching printsign(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Float64, ::Int64, ::Bool, ::MIME{Symbol("text/plain")})                                                                    
``

The j element is no longer being used by the correlated **Polynomials** package code.

Would you mind if I remove the extra argument and push the correction ?

Regards,
André.

mfalt, 2019-10-21:
I noticed that, but forgot to push a fix. Feel free to do it, just make sure to set v0.5.3 as the lower limit for Polynimials.jl

andrersimoes, 2019-10-22:
Hi Mattias.

Do I need any procedure or specific permission to accomplish git push ?

``
remote: Permission to JuliaControl/ControlSystems.jl.git denied to andrersimoes
fatal: unable to access 'https://github.com/JuliaControl/ControlSystems.jl.git/': The requested URL returned error: 403
``
By the way, it seems that your manifest is already up-to-date with Polynomials v0.5.3. Only polyprint required a patch at my side.

baggepinnen, 2019-10-22:
You need to push to your for of Controlsystems.jl and create a pull request to this repository.

phelipe, 2019-10-25:
It was merged?

mfalt, 2019-10-25:
Just merged it @phelipe

closed 2019-10-25 by mfalt

phelipe, 2019-11-12:
I did a pull-request to update project.toml( #237), I think it solve problems with DSP.jl.