JuliaControl/ControlSystems.jl #241: dare method without loading ControlSystems

ueliwechsler, 2019-12-04:
I like to use the `dlqr` method you implemented in my package. I am working on a model predictive control "package" where I need to solve the `dare` to get control gain of the auxiliary controller. 

But this would be the only method needed from `ControlSystems` and the loading time for `ControlSystems` is with 30 plus seconds on my computer rather high. Is there a way to load this method without loading the entire package?

baggepinnen, 2019-12-04:
Yeah, you can copy paste it into your own code 😉

mfalt, 2019-12-05:
Regarding the load time. I did a quick benchmark on my own computer. Usiing ControlSystems takes about 10-12 seconds after reloading julia if everything is precompiled. This is probably a result of these load times:
```julia
@time using Plots          # 3.0s
@time using LaTeXStrings   # 1.6s
@time using LinearAlgebra  # 0.9s 
@time using OrdinaryDiffEq # 4.5-6.5s 
@time using DelayDiffEq    # 0.12s
@time import DSP: conv     # 0.36s
```
We have been working on removing Plots as a dependency, and instead using RecipesBase. This would reduce load times by 3s (i.e almost 30%). However, getting rid of OrdinaryDiffEq and DelayDiffEq would be trickier, and even DiffEqBase takes about 3.5s to load.

baggepinnen, 2019-12-05:
Latency improvements are on top of the priority list for core devs for the
1.4 release, so we can probably expect some free improvements within the
next few months :)

Sent from mobile device

ueliwechsler, 2019-12-05:
Thanks for your replies. 

For now, I would just copy-paste the code and cite `ControlSystems` and then review the load-time situation in some months time. Given that this is okay for you to just copy-paste your code.

mfalt, 2019-12-05:
> Thanks for your replies.
> 
> For now, I would just copy-paste the code and cite `ControlSystems` and then review the load-time situation in some months time. Given that this is okay for you to just copy-paste your code.

Everything in the package is under MIT license, so you are free to use it as you want.

closed 2019-12-06 by ueliwechsler