JuliaControl/ControlSystems.jl #263: Error printing type to REPL

jonniedie, 2020-02-12:
Using Windows 10, errors both in Atom and plain Julia REPL.

```julia
julia> using ControlSystems

julia> s = tf("s")
TransferFunction{ControlSystems.SisoRational{Int64}}
Error showing value of type TransferFunction{ControlSystems.SisoRational{Int64}}:
ERROR: MethodError: no method matching printsign(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Int64, ::Int64, ::Bool, ::MIME{Symbol("text/plain")})
Closest candidates are:
  printsign(::IO, ::T, ::Any, ::Any) where T at C:\Users\jdiegelm\.julia\packages\Polynomials\i0ynH\src\show.jl:126
Stacktrace:
 [1] showterm2(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Polynomials.Poly{Int64}, ::Int64, ::Bool, ::Symbol, ::MIME{Symbol("text/plain")}) at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\SisoTfTypes\polyprint.jl:34
 [2] printpoly2(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Polynomials.Poly{Int64}, ::Symbol, ::MIME{Symbol("text/plain")}) at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\SisoTfTypes\polyprint.jl:19
 [3] printpolydesc at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\SisoTfTypes\polyprint.jl:9 [inlined]
 [4] #7 at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\SisoTfTypes\polyprint.jl:2 [inlined] (repeats 2 times)
 [5] #sprint#339(::Nothing, ::Int64, ::typeof(sprint), ::Function, ::Polynomials.Poly{Int64}) at .\strings\io.jl:105
 [6] sprint at .\strings\io.jl:101 [inlined]
 [7] print_siso(::IOContext{REPL.Terminals.TTYTerminal}, ::ControlSystems.SisoRational{Int64}, ::Symbol) at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\SisoTfTypes\SisoRational.jl:47
 [8] show(::IOContext{REPL.Terminals.TTYTerminal}, ::TransferFunction{ControlSystems.SisoRational{Int64}}) at C:\Users\jdiegelm\.julia\packages\ControlSystems\pipUm\src\types\TransferFunction.jl:191
 [9] show(::IOContext{REPL.Terminals.TTYTerminal}, ::MIME{Symbol("text/plain")}, ::TransferFunction{ControlSystems.SisoRational{Int64}}) at .\multimedia.jl:47
 [10] display(::REPL.REPLDisplay, ::MIME{Symbol("text/plain")}, ::Any) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:132
 [11] display(::REPL.REPLDisplay, ::Any) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:136
 [12] display(::Any) at .\multimedia.jl:323
 [13] #invokelatest#1 at .\essentials.jl:709 [inlined]
 [14] invokelatest at .\essentials.jl:708 [inlined]
 [15] print_response(::IO, ::Any, ::Bool, ::Bool, ::Any) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:156
 [16] print_response(::REPL.AbstractREPL, ::Any, ::Bool, ::Bool) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:141
 [17] (::REPL.var"#do_respond#38"{Bool,REPL.var"#48#57"{REPL.LineEditREPL,REPL.REPLHistoryProvider},REPL.LineEditREPL,REPL.LineEdit.Prompt})(::Any, ::Any, ::Any) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:719
 [18] #invokelatest#1 at .\essentials.jl:709 [inlined]
 [19] invokelatest at .\essentials.jl:708 [inlined]
 [20] run_interface(::REPL.Terminals.TextTerminal, ::REPL.LineEdit.ModalInterface, ::REPL.LineEdit.MIState) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\LineEdit.jl:2306
 [21] run_frontend(::REPL.LineEditREPL, ::REPL.REPLBackendRef) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:1045
 [22] run_repl(::REPL.AbstractREPL, ::Any) at D:\buildbot\worker\package_win64\build\usr\share\julia\stdlib\v1.3\REPL\src\REPL.jl:201
 [23] (::Base.var"#770#772"{Bool,Bool,Bool,Bool})(::Module) at .\client.jl:382
 [24] #invokelatest#1 at .\essentials.jl:709 [inlined]
 [25] invokelatest at .\essentials.jl:708 [inlined]
 [26] run_main_repl(::Bool, ::Bool, ::Bool, ::Bool, ::Bool) at .\client.jl:366
 [27] exec_options(::Base.JLOptions) at .\client.jl:304
 [28] _start() at .\client.jl:460
```

baggepinnen, 2020-02-13:
Which version of Controlsystems.jl are you using, I believe that bug should have been fixed?

mfalt, 2020-02-13:
This should have been fixed in v0.5.4 (2019-11-13). Before that we didn't have dependency bounds on our dependencies, so if you have trouble with updating controlsystems it might be that you have fixed an old version of Plots, DifferentialEquations or similar.

jonniedie, 2020-02-13:
That was it. I was at v0.5.3 and updating was blocked by a dirty registry. I did what was described [here](https://discourse.julialang.org/t/registry-dirty/17989/30) and it works now.

closed 2020-02-13 by jonniedie