JuliaControl/ControlSystems.jl #268: Transfer function not displayable

freemin7, 2020-03-13:
```
tf(Poly([1.0]),Poly([1.0]))
TransferFunction{ControlSystems.SisoRational{Float64}}
Error showing value of type TransferFunction{ControlSystems.SisoRational{Float64}}:
ERROR: MethodError: no method matching printsign(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Float64, ::Int64, ::Bool, ::MIME{Symbol("text/plain")})
Closest candidates are:
  printsign(::IO, ::T, ::Any, ::Any) where T at /home/joto/.julia/packages/Polynomials/XpnvY/src/show.jl:129
Stacktrace:
 [1] showterm2(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Poly{Float64}, ::Int64, ::Bool, ::Symbol, ::MIME{Symbol("text/plain")}) at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/SisoTfTypes/polyprint.jl:34
 [2] printpoly2(::Base.GenericIOBuffer{Array{UInt8,1}}, ::Poly{Float64}, ::Symbol, ::MIME{Symbol("text/plain")}) at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/SisoTfTypes/polyprint.jl:19
 [3] printpolydesc at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/SisoTfTypes/polyprint.jl:9 [inlined]
 [4] #7 at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/SisoTfTypes/polyprint.jl:2 [inlined] (repeats 2 times)
 [5] #sprint#339(::Nothing, ::Int64, ::typeof(sprint), ::Function, ::Poly{Float64}) at ./strings/io.jl:105
 [6] sprint at ./strings/io.jl:101 [inlined]
 [7] print_siso(::IOContext{REPL.Terminals.TTYTerminal}, ::ControlSystems.SisoRational{Float64}, ::Symbol) at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/SisoTfTypes/SisoRational.jl:47
 [8] show(::IOContext{REPL.Terminals.TTYTerminal}, ::TransferFunction{ControlSystems.SisoRational{Float64}}) at /home/joto/.julia/packages/ControlSystems/pipUm/src/types/TransferFunction.jl:191
 [9] show(::IOContext{REPL.Terminals.TTYTerminal}, ::MIME{Symbol("text/plain")}, ::TransferFunction{ControlSystems.SisoRational{Float64}}) at ./multimedia.jl:47
 [10] display(::REPL.REPLDisplay, ::MIME{Symbol("text/plain")}, ::Any) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:132
 [11] display(::REPL.REPLDisplay, ::Any) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:136
 [12] display(::Any) at ./multimedia.jl:323
 [13] print_response(::REPL.AbstractREPL, ::Any, ::Bool, ::Bool) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:141
 [14] (::REPL.var"#do_respond#38"{Bool,Atom.var"#228#229",REPL.LineEditREPL,REPL.LineEdit.Prompt})(::Any, ::Any, ::Any) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:719
 [15] #invokelatest#1 at ./essentials.jl:709 [inlined]
 [16] invokelatest at ./essentials.jl:708 [inlined]
 [17] run_interface(::REPL.Terminals.TextTerminal, ::REPL.LineEdit.ModalInterface, ::REPL.LineEdit.MIState) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/LineEdit.jl:2306
 [18] run_frontend(::REPL.LineEditREPL, ::REPL.REPLBackendRef) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:1045
 [19] run_repl(::REPL.AbstractREPL, ::Any) at /builddir/build/BUILD/julia/build/usr/share/julia/stdlib/v1.3/REPL/src/REPL.jl:201
 [20] (::Base.var"#770#772"{Bool,Bool,Bool,Bool})(::Module) at ./client.jl:382
```
It currently seems impossible to display transfer functions.

baggepinnen, 2020-03-14:
Are you using the latest release of Controlsystems.jl?

freemin7, 2020-03-14:
This issue occurs in
```
  [a6e380b2] ControlSystems v0.5.3
```
which is the latest version i get on Julia 1.3.
Switching to the master branch: 
```
  [a6e380b2] ControlSystems v0.5.6 #master (https://github.com/JuliaControl/ControlSystems.jl.git)
```
fixes it as described in bold above

mfalt, 2020-03-14:
This problem occurs because we didn't have any boundaries on dependencies up until 0.5.3. So, for example, if you have fixed plots.jl to 0.23 or lower, that doesn't satisfy the dependencies. I assume that is why it seems like you are not able to get the newer versions automatically. You do however seem to have a newer version of polynomials.jl which causes this bug. That was one of the reasons why we felt the need to put boundaries on the dependencies.

freemin7, 2020-03-14:
```
(v1.3) pkg> status
    Status `~/.julia/environments/v1.3/Project.toml`
  [ca8b7239] ACME v0.9.2
  [c52e3926] Atom v0.12.9
  [ecbce9bc] BenchmarkProfiles v0.3.1
  [6e4b80f9] BenchmarkTools v0.5.0
  [c3b6d118] BitIntegers v0.2.2
  [764a87c0] BoundaryValueDiffEq v2.4.0
  [7057c7e9] Cassette v0.3.1 #master (https://github.com/jrevels/Cassette.jl.git)
  [3abffc1c] ControlSystemIdentification v0.1.5
  [a6e380b2] ControlSystems v0.5.3
  [a93c6f00] DataFrames v0.20.2
  [864edb3b] DataStructures v0.17.10
  [eb300fae] DiffEqBiological v4.2.0
  [aae7a2af] DiffEqFlux v1.7.0
  [1130ab10] DiffEqParamEstim v1.12.0
  [0c46a032] DifferentialEquations v6.11.0
  [31c24e10] Distributions v0.22.5
  [497a8b3b] DoubleFloats v1.1.3
  [fa6b7ba4] DualNumbers v0.6.2
  [587475ba] Flux v0.10.3
  [d7ba0133] Git v0.3.0
  [4df31cd9] Grassmann v0.5.4
  [cd3eb016] HTTP v0.8.12
  [7073ff75] IJulia v1.21.1
  [e5e0dc1b] Juno v0.8.1
  [d4b2101a] Lint v0.0.0 #master (https://github.com/tonyhffong/Lint.jl)
  [09606e27] ODEInterfaceDiffEq v3.5.0
  [429524aa] Optim v0.20.1
  [1dea7af3] OrdinaryDiffEq v5.29.2 #myb/unit (https://github.com/JuliaDiffEq/OrdinaryDiffEq.jl.git)
  [14b8a8f1] PkgTemplates v0.6.4
  [91a5bcdd] Plots v0.29.7
  [f27b6e38] Polynomials v0.6.1
  [c46f51b8] ProfileView v0.6.3
  [d330b81b] PyPlot v2.8.2
  [1fd47b50] QuadGK v2.3.1
  [276daf66] SpecialFunctions v0.10.0
  [f3b207a7] StatsPlots v0.14.1
  [1986cc42] Unitful v1.0.0
  [228000da] XMLDict v0.4.0
  [e88e6eb3] Zygote v0.4.9
  [ade2ca70] Dates 
```
In case it helps diagnose the problem.

mfalt, 2020-03-14:
Thanks, yes, it is probably "Plots v0.29.7" that is the problem. Although it was released a month ago, we have no robust automatic testing in place to verify that the plots look reasonable, so we have to test that manually, therefore we might be a bit slow in updating our dependencies to include new versions. I assume a `]free Plots` would have allowed you to bump ControlSystems by doing an `]up`. But I have added 0.29 now, so `]up` alone should update ControlSystems to 0.5.7 (https://github.com/JuliaControl/ControlSystems.jl/pull/271) once the new version is tagged merged into the registry.

Looking closer, it could also be that you have manually added "OrdinaryDiffEq v5.29.2" which is not supported by any recent version of DelayDiffEq.jl which we have a dependency on.

All these dependencies are really a mess to keep track of, and I am not sure what the best solution here is.

mfalt, 2020-03-17:
Feel free to re-poen this if you are still having problems.

mfalt, 2020-03-17:
Feel free to re-open this if you are still having problems.

closed 2020-03-17 by mfalt