JuliaControl/ControlSystems.jl #278: Problems with Polynomials again?

B-LIE, 2020-03-31:
In 2019, there were some issues with Polynomials, with commands `tf(sys)`, etc. not working. Then, in January - February of 2020, this worked. But now, it doesn't work again.

I'm still on Julia 1.3... would it help to upgrade to Julia 1.4?

mfalt, 2020-03-31:
What version of ControlSystems.jl are you using? It should not be a problem since v0.5.4, and I don't think the julia version should matter.

B-LIE, 2020-03-31:
I'm on v0.5.3. Could it be that I have installed a package lately that has enforced a *downgrade* of ControlSystems.jl to an older version? It definitely worked in early February.

B-LIE, 2020-03-31:
... here is the entire list of my packages (`Pkg.status()`) after `Pkg.update()`:
[c52e3926] Atom v0.12.5
  [6e4b80f9] BenchmarkTools v0.4.3
  [a134a8b2] BlackBoxOptim v0.5.0
  [336ed68f] CSV v0.5.24
  [861a8166] Combinatorics v1.0.0
  [34da2185] Compat v2.2.0
  [a6e380b2] ControlSystems v0.5.3
  [a93c6f00] DataFrames v0.20.2
  [82cc6244] DataInterpolations v1.3.1
  [864edb3b] DataStructures v0.17.9
  [0c46a032] DifferentialEquations v6.11.0
  [31c24e10] Distributions v0.21.9
  [c87230d0] FFMPEG v0.2.4
  [587475ba] Flux v0.9.0
  [59287772] Formatting v0.4.1
  [f6369f11] ForwardDiff v0.10.9
  [38e38edf] GLM v1.3.7
  [28b8d3ca] GR v0.46.0
  [a2cc645c] GraphPlot v0.4.1
  [9fb69e20] Hiccup v0.2.2
  [7073ff75] IJulia v1.21.1
  [d1acc4aa] IntervalArithmetic v0.16.1
  [138f1668] IntervalConstraintProgramming v0.12.0
  [d2bf35a9] IntervalRootFinding v0.5.2
  [e5e0dc1b] Juno v0.8.0
  [b964fa9f] LaTeXStrings v1.0.3
  [984bce1d] LambertW v0.4.5
  [23fbe1c1] Latexify v0.13.0
  [093fc24a] LightGraphs v1.3.0
  [9c8b4983] LightXML v0.8.1
  [6f286f6a] MultivariateStats v0.7.0
  [2774e3e8] NLsolve v4.3.0
  [0f4fe800] OMJulia v0.1.0 #master (https://github.com/OpenModelica/OMJulia.jl)
  [ab2f91bb] OSQP v0.6.0
  [429524aa] Optim v0.20.1
  [91a5bcdd] Plots v0.29.1
  [f27b6e38] Polynomials v0.5.3
  [08abe8d2] PrettyTables v0.8.4
  [c46f51b8] ProfileView v0.6.3
  [92933f4c] ProgressMeter v1.2.0
  [438e738f] PyCall v1.91.3
  [d330b81b] PyPlot v2.8.2
  [1fd47b50] QuadGK v2.3.1
  [37e2e3b7] ReverseDiff v1.1.0
  [90137ffa] StaticArrays v0.12.1
  [2913bbd2] StatsBase v0.32.1
  [c3572dad] Sundials v3.8.3
  [24249f21] SymPy v1.0.7
  [b8865327] UnicodePlots v1.2.0
  [c2297ded] ZMQ v1.2.0
  [9abbd945] Profile
  [9a3f8284] Random

B-LIE, 2020-03-31:
The only ones I recall I have installed lately are `QuadGK` and `ProgressMeter`.

mfalt, 2020-03-31:
Polynomials is definitely at an old version, 0.6 is required. What happens if you do
]add ControlSystems@0.5.8
Maybe that will show what is blocking

B-LIE, 2020-03-31:
Hm. I did `]add ControlSystems@0.5.8` and was told it couldn't be done because of an old version of `Polynomials`...
Then I did `]add Polynomials@0.6`, and then it updated `Polynomials` to vers. `0.6.1`. After that, I ran `Pkg.update()` again, and now a recent version of both packages appear to be installed. 

When I tried to run my code, I got loads of error messages, though. But after I did `Pkg.build("Sundials")` and `Pkg.build("PyCall")` -- things started to work.

B-LIE, 2020-03-31:
I don't know why `Polynomials` automatically updated to the latest version when I did `Pkg.update()`...

closed 2020-10-01 by baggepinnen