JuliaControl/ControlSystems.jl #304: Unity feedback of discrete-time system

olof3, 2020-05-15:
`feedback(ss(0.5, 1, 1, 0, 1), 1)` currently throws "sampletime mismatch"

I guess that this will be resolved after pulling the sampletime PR, so this is just a reminder to check that this is indeed the case.

Here are two simple regression tests to add.
```julia
@test feedback(ss(0.5, 1, 1, 0, 1), 1) == ss(-0.5, 1, 1, 0, 1)
@test feedback(1, ss(0.5, 1, 1, 0, 1)) == ss(-0.5, 1, -1, 1, 1)
```

olof3, 2020-05-15:
We should support feedback with `UniformScaling`, `1` wouldn't really make sense for MIMO systems. Perhaps still makes sense to support feedback with `Number` for the SISO case?

mfalt, 2020-10-30:
Did we solve this?

olof3, 2020-10-31:
Nope. This is still an issue. I also found this problem

```julia
@test [ss(1,1,1,0,1) 1] == ss(1, [1 0], 1, [0 1], 1)
```

We've had all these problems for systems with different sample times etc. Couldn't we just rely on `promote` to a greater extent than `convert`? It seems like this would make it a lot easier to get the sampletime right.

PR #626 (2022-01-28): Add tests for discrete feedback
closes #304
commit ad6f07e8 (2022-01-28): Add tests for discrete feedback (#626)

closes #304
closed 2022-01-28 by baggepinnen