JuliaControl/ControlSystems.jl #310: General bad performance for plotting.

Kelvyn88, 2020-07-11:
Any suggestion to speed up the performance? 

When plotting root-locus, even stepplots, the package takes a long time to run, more than 5 minutes. 

I use julia 1.4.2 in Win10, i7 8750h, 32 GB Ram.

baggepinnen, 2020-07-11:
Which version of Controlsystems.jl are you using? How long time does it take to load Ordinarydiffeq.jl and solve a simple ode? Both of the methods you referred to make use of Ordinarydiffeq

Kelvyn88, 2020-07-11:
ControlSystems is 0.6.0 and I also try the master branch. 

OrdinaryDiffEq takes few seconds to solve the example provided in the github repository. 

An example of the code that I am running. It takes 555 seconds (sorry for the spanish comments):

```
#retardo
start = time()
using Plots, LaTeXStrings
import ControlSystems, SymPy

pyplot()

const SP = SymPy
const CS = ControlSystems

#Lazo abierto
den = 1.06
nuo3 = 1
nuo2 = 3
nuo1 = 2
nuo0 = 0
Gopen = CS.tf([den], [nuo3, nuo2, nuo1, nuo0])
println("Lazo Abierto")
println(Gopen)
display(CS.impulseplot(Gopen, 12))

#Lazo cerrado
Gclo = CS.feedback(Gopen)
println("Lazo Cerrado")
println(Gclo)
display(CS.impulseplot(Gclo, 12, framestyle = :box))
display(CS.stepplot(Gclo, 12, framestyle = :box))

#Calculo de indicadores de respuesta
Wn, ζ, ps = CS.damp(Gclo)

println("Wn =")
println(Wn[1])
println("ζ =")
println(ζ[1])
println("Polos Lazo Cerrado")
println(ps)

CS.dampreport(Gclo)

#calculo de kv
sl = SP.symbols("sl", real = true)
Greal = den / (nuo3 * sl^3 + nuo2 * sl^2 + nuo1 * sl^1 + nuo0)
Glim = sl * Greal
Kvs = SP.limit(Glim, sl, 0)
println(Kvs)
Kvo = convert(Float64, Kvs)
println("Kv")
println(Kvo)

#datos de diseño
Kvd = 5 #entrada
β = Kvd / Kvo
T = 20 #entrada
println("Kv deseada")
println(Kvd)
println("β")
println(β)
Zero = -(1 / T)
Pole = -(1 / (β * T))
println("Zero")
println(Zero)
println("Polo")
println(Pole)

#FUNCION DEL COMPENSADOR
Gc1 = CS.tf([1, -Zero], [1, -Pole])
println("Funcion del compensador sin Kc")
println(Gc1)
Gce1 = CS.feedback(Gc1 * Gopen)
println("Lazo cerrado sin Kv")
println(Gce1)

Wn, ζ, ps = CS.damp(Gce1)
println("Wn =")
println(Wn[1])
println("ζ =")
println(ζ[1])
println("polo nuevo")
println(ps[2])
CS.dampreport(Gce1)
pd1 = ps[2]
#Con el polo calc angulo

#contribucion de ángulo
s = pd1
Gopenn = den / (nuo3 * s^3 + nuo2 * s^2 + nuo1 * s^1 + nuo0)
println("sustituyendo el polo deseado")
println(Gopenn)
ϕ = atand(imag(Gopenn) / real(Gopenn))
println("ϕ contribución de angulo")
println(ϕ)

#Obtener Kc
Gcer1 = Gopen * Gc1
println("Lazo abierto sin compensador")
println(Gcer1)
Gcreal = ((s - Zero) / (s - Pole)) * Gopenn
println("sustituyendo el polo deseado")
println(Gcreal)

#calculo de kc
Kc = 1 / sqrt((imag(Gcreal)^2) + (real(Gcreal)^2))
println("Kc")
println(Kc)

#calculo del lazo cerrado compensado
Gc2 = Kc * Gc1
println("Funcion del compensador")
println(Gc2)
Gcopen = Gc2 * Gopen
println("Funcion de lazo abierto compensada")
println(Gcopen)
Gclop = CS.feedback(Gcopen)
println("Funcion de lazo cerrado compensado")
println(Gclop)
display(CS.rlocusplot(Gcopen))
display(CS.stepplot(
    Gclo,
    10,
    lw = 1.5,
    lc = :blue,
    label = L"G_{uncompensed\hspace{0.4}system}",
))
display(CS.stepplot!(
    Gclop,
    10,
    framestyle = :box,
    lc = :red,
    label = L"G_{Compensed\hspace{0.4}system}",
    title = "",
    xlims = (0, 10),
    ylim = (0, 2.0),
    yticks = (0:0.4:2.0),
    xticks = (0:1:10),
    lw = 1.5,
    linestyle = :dot,
    legendfontsize = 14,
    guidefontsize = 14,
    tickfontsize = 14,
    background_color_legend = RGB(174 / 255, 50 / 255, 205 / 255),
))

#segundocalculo de kv
Greal2 =
    (den / (nuo3 * sl^3 + nuo2 * sl^2 + nuo1 * sl^1 + nuo0)) *
    ((sl - Zero) / (sl - Pole))
Glim2 = sl * Greal2
Kv2 = SP.limit(Glim2, sl, 0)
println("Kv sistema compensado")
println(Kv2)

#Calculo de resistencias
R1 = SP.symbols("R1", real = true)
R2 = SP.symbols("R2", real = true)
R3 = SP.symbols("R3", real = true)
R4 = SP.symbols("R4", real = true)
C1 = SP.symbols("C1", real = true)
C2 = SP.symbols("C2", real = true)

#Equations from Ogata
C1 = 1E-6
C2 = 1E-7
R3 = 10E3
eqs = [(1 / T) - R1 * C1, (1 / (β * T)) - R2 * C2, Kc - ((R4 * C1) / (R3 * C2))]
e = SP.solve(eqs)
println(e)
endtime = time() - start
```

Kelvyn88, 2020-07-11:
Maybe the low bad performance comes when calling rlocusplot, stepplot, impulseplot and so on.

baggepinnen, 2020-07-13:
Well, this script does quite a lot of things, including using SymPy. The initial plots all appear rather quickly here so I can't really tell you what's wrong. The following taskes 0.3 seconds on my computer
```julia
@time begin
Gopen = CS.tf([den], [nuo3, nuo2, nuo1, nuo0])
println("Lazo Abierto")
println(Gopen)
display(CS.impulseplot(Gopen, 12))

#Lazo cerrado
Gclo = CS.feedback(Gopen)
println("Lazo Cerrado")
println(Gclo)
display(CS.impulseplot(Gclo, 12, framestyle = :box))
display(CS.stepplot(Gclo, 12, framestyle = :box))

#Calculo de indicadores de respuesta
Wn, ζ, ps = CS.damp(Gclo)

println("Wn =")
println(Wn[1])
println("ζ =")
println(ζ[1])
println("Polos Lazo Cerrado")
println(ps)

CS.dampreport(Gclo)
end
0.290803 seconds (76.20 k allocations: 4.201 MiB)
```

Kelvyn88, 2020-07-13:
It is weird, now it is running much faster even with SymPy. Thanks in advanced. 

I was using Atom, maybe that was the problem, and solved with a recent update.

closed 2020-07-13 by Kelvyn88