JuliaControl/ControlSystems.jl #317: Precompilation crash due to Polynomials.jl@1.1.7

NilsVreman, 2020-09-17:
ControlSystems.jl@0.6.0 crashed when I tried to precompile it using the dependency Polynomials.jl@1.1.7.

Precompilation worked after installing a previous version of Polynomials.jl (@1.0.0).

olof3, 2020-09-22:
There has been a lot of problems with Polynomials.jl recently.

But this worked for me with ControlSystems.jl@0.6.0 and Polynomials.jl@1.1.7 on a relatively recent version of Julia. What Julia version did you use (although that shouldn't be a problem)?

NilsVreman, 2020-09-22:
> There has been a lot of problems with Polynomials.jl recently.
> 
> But this worked for me with ControlSystems.jl@0.6.0 and Polynomials.jl@1.1.7 on a relatively recent version of Julia. What Julia version did you use (although that shouldn't be a problem)?

I'm using Julia@1.1.0. I also found it strange.

olof3, 2020-09-22:
Just saw someone with a lot of dependency problems for ControlSystems and Julia@1.2, so using a more recent version could do the trick, or perhaps it is the fresh install of all the packages that does it.

 The tests pass for v1.0 though.

baggepinnen, 2020-10-01:
Any reason you use an old version of Julia? You're missing out on the new threading in 1.3 and latency fixes in 1.4-1.5.
Can this issue be closed or there is some action to be taken?

NilsVreman, 2020-10-01:
This issue can be closed. I will update my Julia version as soon as I have opportunity to update it!

closed 2020-10-01 by NilsVreman