JuliaControl/ControlSystems.jl #329: lsim in julia vs python

aamirmushirkhan, 2020-09-24:
HI, Is there any way through which I can use the lsim function in this toolbox for my python code? For my analysis, the lsim in python is quite slow. So, I am looking for alternatives.

baggepinnen, 2020-09-24:
You can call Julia from Python through the package pyjulia, in practice i think it will be a bit cumbersome since lsim expects Julia types representing the LTI systems and there is no automatic conversion from the corresponding Python types. The best be would probably be to simply call the underlying numerical function that accepts matrices only.

baggepinnen, 2020-09-29:
Closing this as it's not an issue with this package. Feel free to continue the discussion though.

closed 2020-09-29 by baggepinnen