JuliaControl/ControlSystems.jl #335: Change framestyle for pole-zero maps

KronosTheLate, 2020-09-29:
This might be personal preference, but I ALWAYS think that pzmaps look way better with the axis at zero, aka with the argument to plot ```framestyle=:zerolines```. Therefore I would like to propose changing this to the default.

(I have never opened an issue before, so please let me know if this is the wrong forum for such a discussion or if I have otherwise behaved improperly)

baggepinnen, 2020-09-29:
I've never use that option, but I can imagine it would be a good idea. Feel free to open a PR with the change and maybe a before and after picture so others can chime in as well 😃

KronosTheLate, 2020-09-30:
https://github.com/JuliaControl/ControlSystems.jl/pull/336

closed 2020-09-30 by KronosTheLate