JuliaControl/ControlSystems.jl #347: How do I convert from  discrete to continuous time

StevenSiew, 2020-10-07:
I think in Matlab there is a way to convert from discrete time model to continous time model

sysc = d2c(sysd)
sysc = d2c(sysd,method)

Description
example
sysc = d2c(sysd) converts a the discrete-time dynamic system model sysd to a continuous-time model using zero-order hold on the inputs.

example
sysc = d2c(sysd,method) specifies the conversion method.

How do I do it in ControlSystems.jl

KronosTheLate, 2020-10-07:
Judging by the fact that there is a c2d function, but no d2c function (https://juliacontrol.github.io/ControlSystems.jl/latest/lib/constructors), I would guess it is not yet implemented. I don't know how difficult it would be to create the function.

But perhaps you could use the ControlSystemIdentification package (https://github.com/baggepinnen/ControlSystemIdentification.jl) to identify a system, which I guess will be continuous. It is a roundabout band-aid-fix, but I think it should be doable.

So this issue is more like a request to implement a d2c function, I guess.

baggepinnen, 2020-10-07:
We do not have an implementataion of `d2c` at the moment. It can be done in multiple ways:
- The inverse of a ZoH discretization is fairly straightforward, simply applyying the matrix logarithm to the matrix that comes out of the matrix exponential during ZoH.
- The inverse of an Euler discretization is also straightforward since it more or less only transforms the `A` matrix to `I +Ts*A`.

ControlSystemIdentification only identifies discrete systems and in either case, going through an identification algorithm simply to transform the model to continuous domain is not advised.

A PR implementing `d2c` for the most common discretizations would be welcome :)

StevenSiew, 2020-10-07:
I would love to do a PR on d2c but I haven't done control theory since 1995. I would try myself to do

- The inverse of a ZoH discretization is fairly straightforward, simply applyying the matrix logarithm to the matrix that comes out of the matrix exponential during ZoH.

When I have done it, maybe I could post the code here and see.

StevenSiew, 2020-10-08:
This is what I have come up with in a short time. Please have a look and comment on it

```julia
"""`sysc = d2c(sys, method=:zoh)`
Convert the discrete system `sys` into a continous system using
the provided method. Currently only `:zoh` are provided.
Returns the discrete system `sysc`
"""
function d2c(sys::StateSpace,method::Symbol=:zoh)
    if !isdiscrete(sys)
        error("sys must be a discrete time system")
    end
    val_1 = one( eltype(sys.A) )
    if method == :zoh
        A_c = real.(  (val_1/sys.timeevol.Ts) .* log(sys.A)  )
        B_c = inv(sys.A - I) * A_c * sys.B
        C_c = sys.C
        D_c = sys.D
        return ss(A_c,B_c,C_c,D_c)
    elseif method == :foh || method == :tustin || method == :matched
        error("NotImplemented: Only `:zoh` is implemented so far")
    else
        error("Unsupported method: ", method)
    end
end

function d2c(transferfunc::TransferFunction,method::Symbol=:zoh)
    if !isdiscrete(transferfunc)
        error("transferfunc must be a discrete time system")
    end
    return tf(  d2c(ss(transferfunc),method)  )
end
```

StevenSiew, 2020-10-09:
New proposed code

```julia
"""`sysc = d2c(sys, method=:zoh)`
Convert the discrete system `sys` into a continous system using
the provided method. Currently only `:zoh` are provided.
Returns the discrete system `sysc`
"""
function d2c(sys::StateSpace,method::Symbol=:zoh)
    if !isdiscrete(sys)
        error("sys must be a discrete time system")
    end
    val_1 = one( eltype(sys.A) )
    if method == :zoh
        if det(sys.A - I) == 0
            error("Unable to use zero order hold method because det(A - I) == 0")
        end
        A_c = real.(  (val_1/sys.timeevol.Ts) .* log(sys.A)  )
        B_c = inv(sys.A - I) * A_c * sys.B
        C_c = sys.C
        D_c = sys.D
        return ss(A_c,B_c,C_c,D_c)
    elseif method == :foh || method == :tustin || method == :matched
        error("NotImplemented: Only `:zoh` is implemented so far")
    else
        error("Unsupported method: ", method)
    end
end

function d2c(G::TransferFunction,method::Symbol=:zoh)
    if !isdiscrete(G)
        error("TransferFunc must be a discrete time system")
    end
    return tf(  d2c(ss(G),method)  )
end
```

baggepinnen, 2020-10-09:
The code I posted in the PR does handle the case when det(A-I) =0

StevenSiew, 2020-10-10:
I am sorry but I could not make sense of your code which you posted in PR

```
julia> exp([g1.A g1.B; zeros(1,3)]) |> log # This is the inverse of the c2d
3×3 Array{Complex{Float64},2}:
 0.0+0.0im  1.0+0.0im  -9.71445e-17+0.0im
 0.0+0.0im  0.0+0.0im           1.0+0.0im
 0.0+0.0im  0.0+0.0im           0.0+0.0im
```

Instead I came up with this

```julia
"""`sysc = d2c(sys, method=:zoh)`
Convert the discrete system `sys` into a continous system using
the provided method. Currently only `:zoh` are provided.
Returns the discrete system `sysc`
"""
function d2c(sys::StateSpace,method::Symbol=:zoh)
    if !isdiscrete(sys)
        error("sys must be a discrete time system")
    end
    val_1 = one( eltype(sys.A) )
    if method == :zoh
        if det(sys.A - I) == 0
            # Use alternative method for :zoh because det(A - I) == 0")
            A_c = real.(  (val_1/sys.timeevol.Ts) .* log(sys.A)  )
            E = sum([ ((A_c * sys.timeevol.Ts)^k)/factorial(k + 1)  for k = 0:19])
            B_c = (val_1/sys.timeevol.Ts) .* inv(E) * sys.B
            C_c = sys.C
            D_c = sys.D
            return ss(A_c,B_c,C_c,D_c)
        end
        A_c = real.(  (val_1/sys.timeevol.Ts) .* log(sys.A)  )
        B_c = inv(sys.A - I) * A_c * sys.B
        C_c = sys.C
        D_c = sys.D
        return ss(A_c,B_c,C_c,D_c)
    elseif method == :foh || method == :tustin || method == :matched
        error("NotImplemented: Only `:zoh` is implemented so far")
    else
        error("Unsupported method: ", method)
    end
end
```

PR #355 (2020-10-10): Add d2c function
This version of `d2c` is the inverse of `c2d` for ZoH and handles the case when `det(A-I) = 0`
closes #347 
this replaces #349 which does not handle this common case.

```julia
julia> sys = ss([0 1; 0 0], [0;1], [1 0], 0)
StateSpace{Continuous,Int64,Array{Int64,2}}
A = 
 0  1
 0  0
B = 
 0
 1
C = 
 1  0
D = 
 0

Continuous-time state-space model

julia> sysd = c2d(sys, 1)[1]
StateSpace{Discrete{Int64},Float64,Array{Float64,2}}
A = 
 1.0  1.0
 0.0  1.0
B = 
 0.5
 1.0
C = 
 1.0  0.0
D = 
 0.0

Sample Time: 1 (seconds)
Discrete-time state-space model

julia> d2c(sysd)
StateSpace{ [... cut]
closed 2020-10-10 by baggepinnen