JuliaControl/ControlSystems.jl #378: Better integration with MonteCarloMeasurements

jonniedie, 2020-10-22:
It would be really cool to be able to do bode plots and such with systems with uncertainty ribbons. So like:
```
using ControlSystems, MonteCarloMeasurements

s = tf("s")
G = (s + (3..4)) / (s^2 + (1±0.2)s - (5..6))
stepplot(G)
bodeplot(G)
```

jonniedie, 2020-10-22:
I think there is probably a way to handle this with `register_primitive`, but since this needs to handle arrays of particles to make the ribbon plots, I think there needs to be some internal tinkering done to make it work automatically.

mfalt, 2020-10-22:
In general we try to support these kinds of functions, in particular automatic differentiation, however you might have to use GenericLinearAlgebra for some functions.

The bodeplot works for me after removing the following three lines
```julia
        ind       = findall(minx .<= minor .<= maxx)
        minor     = minor[ind]
        minorText = minorText[ind]
```
that calculates the x ticks for the bode plot, after you make sure to
`using LinearAlgebra`
so you get access to a generic eigenval solver.
There seems to be some problem with the differential equations solver for stepplot with the particles, but the following works
```julia
using ControlSystems, MonteCarloMeasurements, GenericLinearAlgebra

s = tf("s")
G = (s + (3..4)) / (s^2 + (1±0.2)s + (5..6))
stepplot(c2d(G, 0.1))
```
This is a good benchmark for for future cases

baggepinnen, 2020-10-23:
My goal is definitely to have MCM work well together with Controlsystems.jl, in fact, my very first applications for MCM was with this package. I have an example file doing some basic stuff here
https://github.com/baggepinnen/MonteCarloMeasurements.jl/blob/master/examples/controlsystems.jl

In general, one needs to use GenericLinearAlgebra for a lot of functionality. The ode solving does work, one can easily solve standard odes with particle stated, if our step does not work, the issue is probably on our side.

baggepinnen, 2022-01-11:
This generally works quite well now, many functions that previously required GLA now works with MCM. The `bodeplot` example works if `ylims` are manually provided, i.e.,
```julia
bodeplot(G, w, ylims=(1e-3,1e3))
```
or if `unsafe_comparisons` is activated.

closed 2022-01-11 by baggepinnen