JuliaControl/ControlSystems.jl #383: Remove LaTeX in plot recipies

KronosTheLate, 2020-10-29:
At least the LaTeX that is not important, like as in the labels for bode plots.

Don't get me wrong, the LaTeX looks good (when it works). But I have two main criticisms of it:
1) When I try to use a backend like plotlyjs for the ability to read data with my cursor, the LaTeX does not render and the result is very bad. In this sense, the LaTeX restricts functionality (interactivity) by making the plots with unrendered LaTeX prohibitively ugly.
2) As mentioned in https://github.com/JuliaControl/ControlSystems.jl/pull/316, an annoying syntax error printing can occur when producing bode plots, which was attributed by @baggepinnen to "appear(s) with the GR backend and I think it's due to how we provide latex strings".

So it seems to me like less LaTeX in the plots would increase backend-compatibility, and also fix some recurring syntax-error printing bug, without paying a large price

As with other open issues of mine, I want to discuss this before taking a stab at implementation.

albheim, 2020-11-17:
Looking at where the string "latex" appears in the repo it seems that it might only be used for the plot labels in the pid plot and for the labels on logaxis. 

I was thinking if this could be solved with unicode since it is just super and subscripts. But the plot labels in the pid plot are letters in subscript which I believe not all exists in unicode. The logaxis is just numbers in superscripts and does indeed exists in unicode.

I gave GR and PlotlyJS a quick try and unicode numbers as superscript seems to work on both of them at least. 

I have used the [UnicodeFun.jl](https://github.com/SimonDanisch/UnicodeFun.jl) package a bit and they expose a `to_latex` function which can take some simple latex code as input and converts it to unicode.

Not sure if this is anything desirable at all, just thought I would add my thoughts on it.

baggepinnen, 2020-11-17:
Latex strings can be created with the `L""` string macro as well, that might be used some places
```julia
clinton ~/.julia/dev/ControlSystems> rg "L\""
src/pid_design.jl
241:    Plots.plot(kp,ki,linewidth = 1.5, xlabel=L"k_p", ylabel=L"k_i", title="Stability region of P, k_d = $(round(kd, digits=4))"), kp, ki
251:    Plots.plot(kp,ki,linewidth = 1.5, xlabel=L"k_p", ylabel=L"k_i", title="Stability region of P, k_d = $(round(kd, digits=4))"), kp, ki

src/plotting.jl
81:            minorText = [" " for t in minorText]#fill!(minorText, L" ")
```

KronosTheLate, 2020-11-18:
> I was thinking if this could be solved with unicode since it is just super and subscripts. But the plot labels in the pid plot are letters in subscript which I believe not all exists in unicode. The logaxis is just numbers in superscripts and does indeed exists in unicode.

I am mostly concerned with the log axis ticks and the labels for the line being plotted.

> I gave GR and PlotlyJS a quick try and unicode numbers as superscript seems to work on both of them at least.

That sounds great! Do you also get unrendered latex for fx a bodeplot when using Plotly/PlotlyJS? Because I do, and I value the interactivity a lot.

Would a possible fix be to add something like
```julia
if Plots.backend() ∈ ["PlotlyBackend", "PlotlyJSBackend"]
   # some UnicodeFun version of what is currently being used
end
```
somewhere in the `getLogTicks` function? Probably where it is already checking for different backends ofc.

I would just love to be able to use PlotlyJS/Plotly as a backend for plots produced with this package, without the horrible unrendered LaTeX. Fixing the ticks would go most of the way in solving the current issue ^_^

albheim, 2020-11-18:
> Latex strings can be created with the L"" string macro as well, that might be used some places

That is true, also found a few places where they are normal strings with manual escaping in them.

> That sounds great! Do you also get unrendered latex for fx a bodeplot when using Plotly/PlotlyJS? Because I do, and I value the interactivity a lot.

Yeah, seems like it varies a bit with latex support between the backends. It even seems like some people have it working for PlotlyJS as discussed [here](https://discourse.julialang.org/t/latex-code-for-titles-labels-with-plots-jl/1967/7) but it seems inconsistent.

I do think latex looks a bit better than unicode for most things, but if it just the exponents for the log-axis I'd say they are similar enough for me. The problem is all the other stuff that is still latex and can't be converted to unicode.

```julia
using Plots, LaTeXStrings, UnicodeFun
text = raw"\alpha^{-10}"
plot(randn(10), ylabel=latexstring(text), xlabel=to_latex(text))
```
[image]

> Would a possible fix be to add something like
>
>```julia
>if Plots.backend() ∈ ["PlotlyBackend", "PlotlyJSBackend"]
>   # some UnicodeFun version of what is currently being used
>end
>```
>somewhere in the `getLogTicks` function? Probably where it is already checking for different backends ofc.

I think the preference is to be rather backend agnostic.

What I did find was that [here](https://discourse.julialang.org/t/plots-plotlyjs-latex/38250/3) they talk about plotlyjs not being able to handle latex unless mathjax was loaded using an extra kwarg when calling plot as documented [here](https://docs.juliaplots.org/latest/backends/#Mathjax). It does not seem that sending this kwarg directly to `bodeplot` or similar works, but maybe it is possible to allow some passthrough keywords to the backend in the recipe defined in ControlSystems? So if you want to use a certain backend you can supply specific keywords for that which are just passed along, and the ControlSystems package can still be agnostic of backend.

KronosTheLate, 2020-11-26:
> I think the preference is to be rather backend agnostic.

I know, but the plot function is already not backend-agnostic

But I just found that one can pass `ticks=:default` in the plot-command, to produce the following plot with plotly:
[image]

and the following with plotlyjs:
[image]

That basically solves this problem for me. So I suggest that this issue can be closed once this option is highlighted in the documentation

closed 2025-01-12 by baggepinnen