JuliaControl/ControlSystems.jl #384: Allow place to handle multi input systems

albheim, 2020-10-30:
Currently `place` only handles SISO systems and it might be nice to try to implement something similar as matlab has where they find a robust set of poles for multi input systems according to [Robust Pole Assignment in Linear State Feedback](https://perso.uclouvain.be/paul.vandooren/publications/KautskyNV85.pdf). Have not managed to understand the algorithm but could try to take a look at it if it is of interest.

baggepinnen, 2020-12-13:
In anticipation of a better method, the following does an okay job
```julia
using ControlSystems, Optim

function generic_place(A, B::AbstractMatrix, desired_poles; max_tries = 20)
    function loss(K)
        e = eigvals(A-B*K)
        sum(abs2.(e-desired_poles))
    end    
    desired_poles = sort(complex.(desired_poles), by=LinearAlgebra.eigsortby)
    local res
    for i = 1:max_tries
        K = randn(reverse(size(B)))
        res = Optim.optimize(
            loss,
            K,
            BFGS(),
            Optim.Options(
                store_trace       = false,
                show_trace        = false,
                show_every        = 1,
                iterations        = 10000,
                allow_f_increases = false,
                time_limit        = 100,
                x_tol             = 0,
                f_tol             = 0,
                g_tol             = 1e-12,
                f_calls_limit     = 0,
                g_calls_limit     = 0,
                ),
                )
        (res.f_converged || res.g_converged || res.x_converged) && break
    end
    res.minimizer, res
end
            
A = randn(3,3)
B = randn(3,1)

desired_poles = [-1,-2,-3]
PK = place(A,B,desired_poles)
OK, res = generic_place(A,B,desired_poles)

@test PK ≈ OK rtol=1e-2
```
Of course, you may add additional penalties to the loss function, encouraging use of potential redundancy to reduce nosie influence etc.

baggepinnen, 2023-05-26:
Closed by #854