JuliaControl/ControlSystems.jl #395: Integration with ModelingToolkit.jl

KronosTheLate, 2020-11-14:
When looking at the documentation for ModelingToolkit.jl, I noticed that the docs contain the following section called ControlSystem:
https://mtk.sciml.ai/dev/systems/ControlSystem/

That got me really excited, as ModelingToolkit seems to me like a very serious and well-produced package. So I was a little disappointed when I found that the contents in that section are very limited, and only contain a single type that is described as "A system describing an optimal control problem. ...". So it is not the same as this package at all. I don't even know what an optimal control problem is, but based on quick googling it seems very related to the contents of this package.

So I was wondering - is there a foreseeable future where this package is well integrated with ModelingToolkit.jl? Would that be a benefit, or desirable at all? I am very curious about what you, the regular contributors to this system, think about such a prospect ^_^

baggepinnen, 2020-11-14:
> So I was wondering - is there a foreseeable future where this package is well integrated with ModelingToolkit.jl? 

That's not at all impossible, but it requires someone to need it enough to implement it. We do mostly cover linear control systems in this package and for those you hardly need MTK, so the benefit of integration is not obvious unless we start considering nonlinear systems, which would infinitely widen the scope of the package.

baggepinnen, 2023-01-05:
We're now in a pretty good situation w.r.t. integration with MTK, see the docs of
- [ControlSystemsMTK.jl](https://github.com/baggepinnen/ControlSystemsMTK.jl)
- https://help.juliahub.com/juliasimcontrol/dev/examples/mtk_control/

closed 2023-01-05 by baggepinnen