JuliaControl/ControlSystems.jl #396: Use of promote_op problematic

baggepinnen, 2020-11-19:
See discussion in https://github.com/JuliaLang/julia/issues/38436#issuecomment-730200157

We should try to rid ourselves of the use of `promote_op` to avoid future situations like this.

mfalt, 2020-11-20:
I agree that we should fix this. It should be fairly straight forward if we use `one(T)` and `zero(T)` for compute the promotion instead. However, the following in `promotion.jl` will be trickier
```julia
function Base.promote_rule(::Type{StateSpace{TE, T1, MT}}, ::Type{T2}) where {TE, T1, MT, T2<:Number}
    NewMT = Base.promote_op(*, MT, T2)
    StateSpace{TE, eltype(NewMT), NewMT}
end
```

olof3, 2020-11-20:
Didn't we say that we should get rid of the Matrix type argument? That would resolve this?

mfalt, 2020-11-20:
I assume we would have a similar problem for the corresponding `HeteroStateSpace`, but we seem to avoid defining promote rules with numbers by calling the constructors in those cases, so MAYBE that is the better workaround.
But we seem to be inconsistent/wrong with the current implementations, e.g in `types/promotion.jl`
```julia
*(sys::ST, n::Number) where ST <: AbstractStateSpace = StateSpace(sys.A, sys.B, sys.C*n, sys.D*n, sys.timeevol)
/(sys::ST, n::Number) where ST <: AbstractStateSpace = ST(sys.A, sys.B, sys.C/n, sys.D/n, sys.timeevol)
```
and I can't see directly if we catch the matrix case anywhere.