JuliaControl/ControlSystems.jl #407: Migrate from Travis to Github Actions

mfalt, 2020-12-04:
Travis no longer provides free CI for open source packages, see https://discourse.julialang.org/t/travis-ci-changes-pricing-plans/49528

I have not been able to figure out how to apply for open source credits, and it might not be a good long-time solution anyway, so we should probably change to use github actions which is still free for open source projects.
If someone wants to try with a PR, there are some code snippets that can probably be used here
https://discourse.julialang.org/t/easy-workflow-file-for-setting-up-github-actions-ci-for-your-julia-package/49765

baggepinnen, 2020-12-04:
I use github CI for most of my repos nowadays, it's very smooth and on the whole better than Travis, so I welcome the change 🙂

imciner2, 2020-12-04:
:+1: this would definitely be a good move. GitHub actions has come along way and I think is now the better choice. It also allows for all 3 platforms (Windows, MacOS and Linux) to be tested at once, so there is no more need to have separate CI services for different platforms.

I have a few workflows I use for Julia packages that I can put into an MR if you are interested.

mfalt, 2020-12-05:
@imciner2 that would be great. I would have done it myself if I want busy finishing my thesis right now.

PS: I got a response from Travis staying that they don't even have the system for OSS and education projects in place yet, so I don't think we should depend on them in the future.

albheim, 2020-12-06:
Sounds like a good idea to me.

I took a look at the link you posted @mfalt and saw that their sample CI seems to have test and doctest split into 2 different parts, where test is run on 1, 1.x and nightly while doctest is only run on 1. Do we prefer to split it up like this also and remove the call to doctest from runtests.jl? I'm not sure if we have anything specific in the documentation that we want to run on all versions?

albheim, 2020-12-07:
My PR is just me testing out github actions, and might not lead to something working. So if @imciner2 had some nice workflows I did not mean to steal the issue :laughing:

baggepinnen, 2021-01-17:
Done :)

closed 2021-01-17 by baggepinnen