JuliaControl/ControlSystems.jl #415: Custum labels for bodeplot

KronosTheLate, 2020-12-18:
I just want to show the current behaviour first of all:
## GR without comma in label-array: 
[image]
## GR with comma in label-array: 
[image]

## PlotlyJS without comma in label-array:
[image]
## PlotlyJS with comma in label-array:
[image]

None of this is what I want - two labels in the legend, according to the traces, colour coded. Am I doing something wrong? Can this be improved by the plotting recipies?

baggepinnen, 2020-12-18:
I have solved it on my local fork, just need CI to work before I can start submitting PRs, until then, set every other label to an empty string

KronosTheLate, 2020-12-18:
> I have solved it on my local fork, just need CI to work before I can start submitting PRs, until then, set every other label to an empty string

Amazing. Thank you so much for the fast reply, the permanent upcoming fix, and the band-aid fix. All much appriciated <3

closed 2020-12-18 by KronosTheLate