JuliaControl/ControlSystems.jl #419: Stable system with unstable step response

KronosTheLate, 2021-01-09:
So I had this issue a while back, and decided to see if it presists. It does.

My transfer function `TF` is a simple RLC circuit, where the output is the voltage over the capacitor. The bodeplot looks right, and the PZmap and margin plots confirm that it is a stable system. But the step response is exponential. Am I misunderstanding something, or should this be a simple oscillatory response for a short time, followed by a steady-state reference value?

For other examples previously tested, the value went towards a finite output, but it had spikes and irregulatiries forever. I can not create a single example that has a steady output when given a step, other than `TF=tf(1, 1)`. Screenshots of the issue found below:
[image]
[image]
[image]
[image]

baggepinnen, 2021-01-10:
I can reproduce your results since you didn't post any code, but I would check the result of converting the tf to a statespace model. Maybe there has been numerical issues in the conversion due to the small coefficients.

KronosTheLate, 2021-01-10:
All the code is seen in the screenshots, but I have copied it below for copy/pasting:
```
begin
	using ControlSystems, Plots, PlutoUI
	plotlyjs()
	default(:size, (670, 600))
	s = tf("s")
	frac(num, den) = /(num, den)
	oplus(args...) = 1.0/sum(x->1.0/x, args)
	md"Preamble hideout"
end
R, L, C = 1, 4.7e-6, 1.5e-8
TF = frac(1, s^2*L*C+s*R*C+1)
marginplot(TF, ticks=:default)
pzmap(TF)
stepplot(TF)
```

olof3, 2021-01-10:
Transfer function representations are inherently numerically sensitive and I guess it is generally advisable to stay away from them expect for educational purposes.

The conversion from `StateSpace` to `TransferFunction` could certainly be improved to yield a more balanced `StateSpace` representation (the problem is essentially that `c2d(ss(TF), 0.07)` gives a discrete-time system with a zero A-matrix, due to the poor balancing of `ss(TF)`), but I guess that no one has really been too interested in spending that time due to the inherent problems with `TransferFunction` representations.

KronosTheLate, 2021-01-10:
So you guys are saying that the problem has got to do with TransferFunction -> StateSpace? Because I have been using this package for my whole semester, and this problem only occured a few weeks ago. It worked just fine in the preceeding months. So in that cause, it seems to me like some update in the last 1 or 2 months have caused the conversion to be worse.

olof3, 2021-01-10:
Hmm, so *exactly* the same thing used to work a couple of month ago?

Could also be an update of Julia's matrix exponential, could be changes in the `c2d` function, could be something else.

I think problems with these sort of things are to be expected, but I think some sort of fix for getting your example to work should be considered, it is not that poorly conditioned for it to break this badly.

baggepinnen, 2021-01-10:
How does matlab handle this case? Anyone having matlab handy to try it out?

olof3, 2021-01-10:
Had a quick look. The conversion TransferFunction -> StateSpace results in a well-balanced realization.

baggepinnen, 2021-01-10:
Maybe we would default to balancing the ss upon conversion from tf? What balancing would make sense? We can use balreal due to frequent numerical problems, but perhaps `balance_statespace`?

KronosTheLate, 2021-01-10:
> Hmm, so exactly the same thing used to work a couple of month ago?

I have used this package for the last half year without problem (And thank you guys so much for it <3). It is just recently that the stepplot, and only it, started giving me trouble.

In case it is usefull, I want to mention that creating the stepplot gives me the following warning
`┌ Warning: Interrupted. Larger maxiters is needed.
└ @ DiffEqBase C:\Users\densb\.julia\packages\DiffEqBase\znIav\src\integrator_interface.jl:328`

olof3, 2021-01-10:
>Maybe we should default to balancing the ss upon conversion from tf? What balancing would make sense? We can't use balreal due to frequent numerical problems, but perhaps balance_statespace?

No idea what makes the most sense, but any sort of balancing would probably be better than nothing. It would probably be even better to be more clever when choosing the realization, but again I don't know what the best approach is. Probably someone has already thought about this, but didn't find anything after 1 minute googling.

It could be worthwhile to treat the second order case separately if we don't manage a better fix. For higher-order systems one should really, really stay away from transfer function realizations.

albheim, 2021-01-10:
Found a problem with `_default_dt` as discussed in #420 that might affect this. I think it will choose a way too big timespan for this task to simulate. 

Checking matlab it simulates 5e-5 seconds and we simulate 7 seconds. If you remove the zero I talk about in  the other issue this would also return close to 5e-5 seconds as the largest value in the timevector. Does not seems to completely solve the problem, the plot is still not really similar to matlabs. But it looks better and I don't get the warning I got with current version.

olof3, 2021-01-10:
Yes, 0.07 is much to long. So yes, that is definitely something that needs to fixed for this to work. Perhaps it is the most crucial problem.

baggepinnen, 2021-01-11:
@olof3  It appears as if someone (you) has thought of this before
https://github.com/JuliaControl/ControlSystems.jl/blob/master/src/types/conversion.jl#L124
I did some simple tests and it appears as if matlab is doing something quite similar to `balance_statespace`, so we could just remove the comment on the line above and get some better numerical properties.

olof3, 2021-01-11:
Sounds good. But I guess it really is that the scaling of the controllable canonical form that should be tweaked, perhaps so that the state have roughly the same size/gain (in a very loose sense). Which I guess pretty much corresponds to `balance_statespace`.

olof3, 2021-01-23:
So with @albheim 's changes for `_default_dt` we are getting something that looks a bit better. There is another problem that the final time is too short (which is another issue).

However there are still problems in that DifferentialEquations isn't too happy about the poorly conditioned dynamics.
```julia
G = tf(1, [1e-13, 1e-8, 1])
# Compare the following
stepplot(G, 1e-4)
stepplot(c2d(ss(G), ControlSystems._default_dt(G)), 1e-4)
stepplot(ControlSystems.balance_statespace(ss(G))[1], 1e-4)
```

So there seems to be a few ways to fix it:
1) Balance the statespace realizations resulting from tf -> ss
2) Discretize and use discrete-time solver
3) Submit issue to DifferentialEquations.

I think we should do 1) either way. Perhaps 2). I am somewhat surprised that DifferentialEquations didn't handle it better, after all the system has only 2 states.

HablyA, 2021-01-28:
> How does matlab handle this case? Anyone having matlab handy to try it out?

This is the result of Matlab
[image]

PR #623 (2022-01-28): balance tf -> ss by default
Avoinds many problems of porrly conditioned numerics

Closes #419 (I tried the simulation as well and it was no problem with this PR)
closed 2022-01-28 by baggepinnen