JuliaControl/ControlSystems.jl #420: Function `_default_dt` always returning the same value

albheim, 2021-01-10:
This code seems to always return 0.07 for a stable continuous system as input and the villain is row 271 where we take the minimum of a bunch of absolut values and 0, where 0 is always less or equal to an absolute value. It seems to be some simple heuristic to get a time step time based on the poles, and everything but the zero seems reasonable to me. 

Looking back in history I see that this row used to be the same but without the zero, which was then added 5 years ago by @mfalt in https://github.com/JuliaControl/ControlSystems.jl/commit/f53b48904de64e97cb2a575fcc5b78706acb3679. Instead of just removing it I thought I would check if you remembered anything, I assume there was some reason behind the change just that it went wrong? Or am I completely missing something here?

https://github.com/JuliaControl/ControlSystems.jl/blob/6c54a259980fc3ac784b45db395edfcd9ea59300/src/timeresp.jl#L264-L277

mfalt, 2021-01-10:
`minimum` throws an error on an empty collection I'm pretty sure, that's why an extra number is added. One solution is probably to change 0 to Inf in the condition, and then `if r == 0 || r == Inf` since we still need some default when there is a pole at 0.

albheim, 2021-01-10:
> `minimum` throws an error on an empty collection I'm pretty sure, that's why an extra number is added. One solution is probably to change 0 to Inf in the condition, and then `if r == 0 || r == Inf` since we still need some default when there is a pole at 0.

Ahh, that makes sense. I can put together a PR for that. Should I add some test for that? I mean we probably do run tests that use `_default_dt` but apparently we don't test for this at least.

mfalt, 2021-01-10:
Yes, a test would be great, something i should have added when i fixed it.

albheim, 2021-01-10:
> `minimum` throws an error on an empty collection I'm pretty sure, that's why an extra number is added. One solution is probably to change 0 to Inf in the condition, and then `if r == 0 || r == Inf` since we still need some default when there is a pole at 0.

As I mention in the PR, do we really need the `r == 0` case though? Since we know `isstable(sys) == true` in the else clause we know that the real part of all poles is less than 0.