JuliaControl/ControlSystems.jl #422: bode() returns only phases in interval +/- 180°

stephans3, 2021-01-15:
If I want to determine the phase at some frequency using bode(G, w) with w as a 1-element vector, bode() only returns values in the interval (-180°, 180°). 

This is a problem in case of higher-order systems or delay systems with phases below -180°. 

This problem does _not_ occur for vectors w with n>1 elements.

So is it a bug or a feature? :D

`G = tf(1, [1, 3, 3, 1])`
`bode(G, [1.7])  # ϕ = -178°`
`bode(G, [1.78]) # ϕ = 177° is wrong, should be ϕ=-360° + 177° = -183°`

olof3, 2021-01-15:
This was discussed in #312, and I think the feeling is that it would be good to fix it, but #312 requires some more further work for MIMO and delay systems. The current behavior is at least consistent.

baggepinnen, 2021-01-15:
Have you seen the function `unwrap`? I think you just need to `unwrap` the entire phase curve
```julia
bodeplot(G)
```
will plot the phase correctly

olof3, 2021-01-15:
Right, I was mixing things up. However, `unwrap` will not by default give you what you want if there are more than one or two integrators.

baggepinnen, 2021-01-15:
Nope, the problem is still there if the phase starts at below -180 degrees (maybe already below -90?).

mfalt, 2021-01-15:
Yes, the general problem is that it is tricky to include where the bode plot stars. Thus `bode` assumes that the first frequency in the vector is "small enough", and "assumes" that the frequency there is in (-180,180). It will then unwrap from that frequency. That is why you see the problem when you only input one frequency.

baggepinnen, 2022-07-03:
Unwrap also fails if the change in frequency between two grid points is too large
```julia
w = exp10.(LinRange(-2, 5, 200))
P = tf(1,[1,1])*delay(1)
bodeplot(P, w)
```
[image]

baggepinnen, 2025-01-11:
`bodepot` now adjusts the phase at the start based on the number of integrators.
- #950 

As Mattias said above, it relies on the lowest frequency in the grid is small enough, if this does not hold I think it's hard to solve in general. I'll close this issue for now, feel free to continue the discussion if you have new ideas

closed 2025-01-11 by baggepinnen